metadriverse/metadrive #115: Connected Car Scenario

BedirhanKeskin, 2021-11-07:
Hello, 

I want to test connected car scenarios in which multiple agents communicate with each other (V2V). Basic data of vehicles are transmitted such as speed and location etc.  I know your project is for RL but I find it very interesting because it is lightweight. 
My questions are: 

1.  Can I control multiple agents synchronously such as their respawn location, behaviors?
2.  Do you have any limit for the number of agents? 
3.  Is it possible to have an interface between MetaDrive and Simulink? I may need to produce some data using simulink.

Thanks,

QuanyiLi, 2021-11-08:
Hi,

Thanks for your interest. MetaDrive can satisfy most of your requirements. 

First, MetaDrive can extract all information from the scene and fill them into a dict after execution one simulation step. This dict will be a return value of step function: ```o, r, d, info = env.step({action_dict})```. You can overwrite ```get_step_return()``` to retrieve more information and put them into the return dict ```info```.

Second, It is easy to control more than one vehicle simultaneously. The ```env.step()``` function takes```action_dict``` as the input, which are pairs of agent names and agent actions. The simulator engine core will implement the actions for all agents simultaneously. Besides, the number of agents is unlimited, but it would be better to control no more than 50 agents to ensure an acceptable FPS (30-60 FPS for turning on render, 60-120 for offscreen training)

Third, MetaDrive currently doesn't support co-simulating with simulink, while you can try it if simulink provides python interface.

closed 2021-11-08 by BedirhanKeskin

BedirhanKeskin, 2021-11-15:
Hello QuanyiLi, 

I am a newbie to simulators. I just focus on map creation and controlling multiple agents. My project is not about AI and RL. 
 
- How can I create a map which has no end? I mean I want to connect start and end blocks to create a small city map. 
- Is it possible to define the direction of sockets between roadblocks? 
- Where do you assign/store map after it is created using the PG algorithm? Maybe, I can just try to manipulate it. 

 Thanks,

QuanyiLi, 2021-11-16:
Hi, BedirhanKeskin

It is feasible to create a city by connecting blocks one by one and run agents on it.
 
- At first, I will answer your third question for briefly introducing the data structure. You can find two class variables ```self.blocks=[]``` and ```self.road_network=RoadNetwork()``` in ```class BaseMap```, where we store created blocks and global road network for actuating traffic. All class methods are actually for manipulating them for building a map.

- For creating a no-ending map, you can insert new blocks to the end of ```self.blocks```. The way to append a new block can be found in ```class BIG``` where we sample new blocks by ```def sample_block()``` and add it to the world by ```def _forward()```. There are also some minimal runnable demos showing how to combine several blocks in the *tests/vis_block* directory.

- It is impossible to define the direction of sockets since the direction is automatically calculated and determined according to the socket of the last block.

- For city generation, you have to calculate the shape of each block and finally connect them into a circle. A small town demo can also be found in *tests/vis_block/vis_a_small_town.py*.

Don't forget to pull the main branch. I fixed something recently. Good luck!