metadriverse/metadrive #123: Suggestion to run multiple instance in parallel?

breakds, 2021-12-08:
First of all I would like to express my gratitude on this great project. I really like the feature-rich and lightweight nature of MetaDrive as a driving simulator for reinforcement learning.

I am wondering what is the recommended way to run multiple MetaDrive instances in parallel (each one with a single ego-car agent)? This seem to be a common use case for reinforcement learning training. I am currently running a batch of MetaDrive simulator with each of them in wrapped in a process, which does seem to have overheads of extra resource and communication/synchronization. 

Another problem I encountered when running multiple instances (say, 60 instances on single machine) in their own process is that I will get a lot of warning like this:

```
ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection terminated

ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection terminated

ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection terminated

ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection terminated

ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection terminated

ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection terminated

ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection terminated

ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection terminated

ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection terminated
```

I guess this has something to do with audio. This happens even though I am running the `TopDown` environment, which should not involve sound. Did you see those warning when running multiple instances as well?

Also, is there a plan to have a vectorized batch version?

Thanks!

QuanyiLi, 2021-12-09:
Hi Yang,

Thanks for your interests and this valubale question. 

First, we usually train RL agents through [ray/rllib](https://github.com/ray-project/ray) like what we did in ```metadrive/examples/train_generalization_experiment.py```. With ray, we can utilize more than 100 workers to do parallel PPO training  and get the generalization results in 12 hours. Also, we tested the ```SubprocVecEnv``` from stabel_baselines3, and verified that it is compatible with MetaDrive. 

Due to the limitation of [Panda3D](https://www.panda3d.org/), MetaDrive can only have **one instance** in each process. However, it is still feasible to use ray and stable_baselines3, since the two ways both use **multiprocess** instead of **multithread** to wrap each RL environment (worker).

Going back to your question, you should first make sure that your parallel environment wrapper uses **multiprocess** to sample batch. And then check if your machine limits the number of processes. Actually, we recommend to use ray to do large-scale RL training.

As for the warnings, we actually never encountered similar problems, and already disabled all audio related functions. You can try run an simple MetaDrive example on your machine to see if the warnings still appear, and give us new feedback. We will try our best to help solve it.

Quanyi

breakds, 2021-12-09:
Thanks a lot for the detailed and prompt reply, Quanyi! I was using another RL framework but yes I did find the problem running in thread and used process wrapper instead. I am aiming to run 100 workers as well :)

The warning happens when I have **more than 20 workers**. The training went on well even with the warning though.

My understanding is that for `TopDownMetaDrive` and during training, we do not need Panda3D since it is for rendering. Is that right?

QuanyiLi, 2021-12-10:
Great! I am glad for your progress. 

And your understanding is almost right. The rendering module and graphics module of Panda3D are disabled for saving resources in ```TopDownMetaDrive``` and other environments without camera observation. In this case, Panda3D only provides MetaDrive with the built-in [Bullet](https://pybullet.org/wordpress/) physics engine which performs efficient collision and contact detection.

breakds, 2021-12-10:
> Great! I am glad for your progress.
>
> And your understanding is almost right. The rendering module and graphics module of Panda3D are disabled for saving resources in `TopDownMetaDrive` and other [...]

Thanks for helping clarifying it!

closed 2021-12-10 by breakds