metadriverse/metadrive #128: Quick Question about Environment

sihwa-park, 2021-12-20:
First of all, thank you for sharing your amazing work!
One quick question. Is there any feature related to traffic signals?
I want to train multiple agents in a more realistic setting.

QuanyiLi, 2021-12-20:
Hi, Sihwa,

Thanks for your interest and valuable feedback. We currently focus on solving challenges raised by unprotected intersections, and thus the traffic light is not implemented. After receiving several feedbacks, however, we plan to add traffic Light to MetaDrive in the next version. It will be released in two months with more features, such as training agents on real maps collected by Waymo.

Quanyi

sihwa-park, 2021-12-21:
Thanks! I will look forward to your update.

closed 2021-12-21 by sihwa-park