metadriverse/metadrive #130: Tracking surrounding vehicles

jayabrata97, 2021-12-24:
Is there any way to track surrounding vehicles' positions and velocities for consecutive time frames? Can we get tracking information from state vector representation in observation?

QuanyiLi, 2021-12-24:
Hi, Joy,

From your description, I assume that you want to construct a new observation with a custom data structure and thus a custom observation space.  Therefore, I would first recommend you to quickly go through our implementations in [metadrive/obs/state_obs.py](https://github.com/decisionforce/metadrive/blob/main/metadrive/obs/state_obs.py) so that you will know how to derive a new class for your special purpose.

For getting surrounding vehicles' information, you can find a method in the same file named **vehicle.lidar.perceive(vehicle)**. It takes the ego vehicle as input and returns cloud points and surrounding objects, in which you can get all surrounding vehicles through filtering the returned surrounding objects. After that, you can access all information about the vehicles, since they are all [BaseVehicle](https://github.com/decisionforce/metadrive/blob/main/metadrive/component/vehicle/base_vehicle.py).

Feel free to contact us, if there are additional questions. And wish you a wonderful holiday : )

Quanyi

pengzhenghao, 2021-12-26:
Hi @jayabrata97 ,

For your information, we already prepare the basic functionality to support retrieving surrounding vehicles information.

You can add the basic information of nearest 4 vehicle to the ego vehicle through into the state observation through:

```python
MetaDriveEnv(config={"vehicle_config": {"lidar": {"num_others": 4}}})
```

[image]

This functionality is implemented in the `get_surrounding_vehicles_info` function of `metadrive/component/vehicle_module/lidar.py`. It uses lidar to get closest vehicles' name and retrieve their information as:

[image]

So the relative distance / velocity of neighbors can be collected and concatenated into the state observation.

jayabrata97, 2021-12-26:
Thanks @QuanyiLi and @pengzhenghao for sharing information.

closed 2022-02-20 by jayabrata97