metadriverse/metadrive #147: 21 dimensions of state vector in `drive_in_single_agent_env.py --observation rgb_camera`

jayabrata97, 2022-02-20:
Hi, @pengzhenghao  can you please tell me how to know what are 21 dimensions of state vector?

jayabrata97, 2022-02-20:
6 dimensions are coming from `ego_state_obs_dim` of `metadrive/obs/state_obs.py`. 10 dimensions are coming from `ego_state_obs_dim` of `metadrive/componet/vehicle_navigation_module/base_navigation.py`. 3 dimensions are coming from `get_line_detector_dim()` of `metadrive/obs/state_obs.py` and last 2 dimensions are coming for `random_agent_model` config of `metadrive/obs/state_obs.py`. Please tell me if I am doing some mistake.

QuanyiLi, 2022-02-20:
Hi jayabrata,

it looks correct.

jayabrata97, 2022-02-20:
Thanks @QuanyiLi . Another question: For example I am using `python drive_in_single_agent_env.py --observation lidar`. Then for reward calculation the whole state vector of 261 dimensions are required or reward can be calculated using  first 21 dimensions of `StateObservation()` only, without using `LidarStateObservation()` also?

QuanyiLi, 2022-02-20:
Good question. 

For reward calculation, some papers may write it as ```r=(o, a)```, but in MetaDrive, the reward is calculated by using other internal states which can not be accessed by the agent, e.g. positions. 
Therefore, the reward can not be calculated by solely using observations, let alone a part of observation.

To better understand how we shape the reward, the reward definition of MetaDrive can be found in our paper: [https://arxiv.org/abs/2109.12674](https://arxiv.org/abs/2109.12674).

closed 2022-05-01 by jayabrata97