metadriverse/metadrive #151: Reproducibility Problem

BedirhanKeskin, 2022-03-07:
Hello, 

I am trying to create custom scenarios. For that, I created a custom map similar to vis_a_small_town.py, and I am using the drive_in_multi_agent_env.py example. The environment is defined as follow: 
`env = envs[env_cls_name](
        {
            "use_render": True, # if not args.pygame_render else False,
            "manual_control": True,
            "crash_done": False,
            #"agent_policy": ManualControllableIDMPolicy, 
            "num_agents": total_agent_number,
            #"prefer_track_agent": "agent3",
            "show_fps": True, 
            "vehicle_config": {
                "lidar": {"num_others": total_agent_number},
                "show_lidar": False,    
            },
            "target_vehicle_configs": {"agent{}".format(i): {
                    #"spawn_lateral": i * 2,
                    "spawn_longitude": i * 10,
                    #"spawn_lane_index":0,
                    "vehicle_model": vehicle_model_list[i],
                    #"max_engine_force":1,
                    "max_speed":100,
                }
                                           for i in range(5)}
        }
    )`

I have a list that consists of steering and throttle_brake values. I am assigning these values regarding step number into env.step: 

`o, r, d, info=env.step({
                'agent0': [agent0.steering, agent0.pedal], 
                'agent1': [agent1.steering, agent1.pedal],
                'agent2': [agent2.steering, agent2.pedal],
                'agent3': [agent3.steering, agent3.pedal],
                'agent4': [agent4.steering, agent4.pedal],
                'agent5': [agent5.steering, agent5.pedal],
                'agent6': [agent6.steering, agent6.pedal],
                'agent7': [agent7.steering, agent7.pedal],
                'agent8': [agent8.steering, agent8.pedal],
                'agent9': [agent9.steering, agent9.pedal],
                }
            )`

At the end of the list for vehicle commands, the counter for loop set the zero, and commands are reused. Also, vehicles' locations, speeds, and headings are set to initial values stored in the dictionary: 

`def initilize_vehicles(env):

    global total_agent_number

    for i in range (total_agent_number):
        agent_str = "agent" + str(i)
        env.vehicles[agent_str].set_heading_theta(vehicles_initial_values[agent_str]['initial_heading_theta'])
        env.vehicles[agent_str].set_position([vehicles_initial_values[agent_str]['initial_position_x'],vehicles_initial_values[agent_str]['initial_position_y']]) # it is x,y from the first block of the map
        env.vehicles[agent_str].set_velocity(env.vehicles[agent_str].velocity_direction, vehicles_initial_values[agent_str]['initial_velocity']) `

 I want to reproduce the scenario and test my main algorithm. However, the problem is vehicles do not act in the same way in every run of scenario. I checked my commands for vehicle using: `env.vehicles["agent0"].steering,env.vehicles["agent0"].throttle_brake` 
Vehicle commands are the same for each repetition of scenarios.  

When I don't use a loop and start the MetaDrive from the terminal, I mostly see the same action from cars. I tested almost 10 times. But in loop case, cars start to act differently after the first loop. 

Reproducibility is a huge concern for me. Is it something about the physic engine? Are there any configuration parameters for the engine? 

Thanks!!

pengzhenghao, 2022-03-07:
Hi @BedirhanKeskin ,

Reproducibility is a huge concern for us too!

You said:

> When I don't use a loop and start the MetaDrive from the terminal, I mostly see the same action from cars. 

That means your environment and MetaDrive are both working well since the first initialization is reproducible!

> I tested almost 10 times. But in loop case, cars start to act differently after the first loop.

That's the problem! The very first loop is deterministic and the second loop starts to show different behavior right?

My first instinct is that the vehicles are not properly reset.

We have a complex logic to reset vehicles especially in the MARL environments. Because the recycle and reallocation of vehicles are the keys to the efficiency when many many vehicles are operating together.

We have an AgentManager class used to manage all controllable vehicles in `metadrive/manager/agent_manager.py`.
In this file, we have a function `def finish(agent_name)` that records which vehicle is done at certain step and processes this. You can refer to the implementation of AgentManager to see the whole workflow.

You might find the above solution to be too detailed. Another easy way to make sure everything is reset well is to call the reset function of the environment: `env.reset()`. In this function, MetaDrive will automatically reset everything in the environment, including calling the reset function of the AgentManager (you can find this function in the `agent_manager.py` too!)

Conclusion: Remember to call `env.reset(...)` first and then call your `initialize_vehicles(env)`, no matter you are in the first loop or second loop!

Please feel free to post anything if my instinct is wrong!

Please also add comment if @QuanyiLi has any suggestion.

QuanyiLi, 2022-03-08:
Hi BedirhanKeskin,

Would you mind providing a script for us to reproduce it? Apart from what was mentioned by Zhenghao, it might be some problems related to the random seed engine. 

Thanks,
Quanyi

BedirhanKeskin, 2022-03-09:
Hello,
@pengzhenghao , I added the env.reset(). But it couldn't solve the problem. After testing more, I saw the problem also when I restart the MetaDrive. So I guess it is not about env.reset().

@QuanyiLi . Would you give me your email? Then I can send my all code. Because I have changes in different modules. 

Thanks for the quick response !!

QuanyiLi, 2022-03-09:
Hi, BedirhanKeskin,

Please send it to *quanyili0057@gmail.com*, I will figure out the problem as soon as possible.

Thanks,
Quanyi

QuanyiLi, 2022-03-11:
Hi, BedirhanKeskin,

I can not reproduce the problem.

I run your code with the following small modification.
[image]

After that, the environment will be reset per 500 steps, and I find that every episode is actually the same. See this video: https://drive.google.com/file/d/1TFZtYEJrccacs4k3vt8FD12ombJRWxJL/view?usp=sharing

So I would like to know if the way I used to reproduce the bug is correct.

Also, it is exciting for us to see what you have done with MetaDrive. It is realizing that it indeed helps others that motivates us to make MetaDrive better. So we'll try our best to support you.

Quanyi

BedirhanKeskin, 2022-03-11:
Hello @QuanyiLi , 

Thanks for the support !! The modification is fine.  I was trying to reset the scenario when the list end but you set a hard-coded value. Because I have scenarios in different sizes. 

The deviation from the correct route becomes obvious when you run the scenario longer.  So, Could you make the value 1600 instead of 500?  I think you will see the problem in a way. 
I am having the same problem in an also short scenario that has 100 steps. I am attaching the files for that scenario. You need to just change the attached files with your ones.  They are in the connectedCar folder.  Don't forget to change their file type to python again :) 

Best Regards,

[map_creator.txt](https://github.com/decisionforce/metadrive/files/8231370/map_creator.txt)
[scenarioConfiguration.txt](https://github.com/decisionforce/metadrive/files/8231371/scenarioConfiguration.txt)

Bedirhan Keskin

QuanyiLi, 2022-03-11:
Hi Bedirhan,

I already figure out why this happen. Previously, we introduce a new optimization to improve the efficiency of ```reset()``` function, where the instantiated BaseVehicle Object is destroyed in a fake way. Actually, these vehicles are still stored in the memory and wait for reusing instead of being released.

As shown in the figure below, you can set the ```force_destroy=True``` instead of ```False``` to disable this optimization. Then vehicles can reproduce trajecctories.
[image]

It is a critical bug but really hard to find lol. Thanks for your feedback!

Quanyi

QuanyiLi, 2022-03-13:
Hi @BedirhanKeskin,

I'd like to know if this solution works for your project : )

Quanyi

BedirhanKeskin, 2022-03-13:
Hello @QuanyiLi ,

I tested like 10 times around 20 minutes and I didn't see any problem.
 
I really appreciate your support!!

closed 2022-03-13 by BedirhanKeskin