metadriverse/metadrive #154: Low fps

Taoismer, 2022-03-12:
I ran the simulator on the remote host, the render fps is only 2.2 fps, which make the simulator unavailable

QuanyiLi, 2022-03-13:
Hi, Taoismer

Previously, some users also reported the low fps problem due to GPU problem, please see #100. Could you please check if MetaDrive is using GPU when you run it on the remote machine? Just use ```watch nvidia-sim```, it is supposed to see the MetaDrive process in the list.

Quanyi

Taoismer, 2022-03-13:
@QuanyiLi No, there is no any GPU occupied. I don't knoe how to solve it

QuanyiLi, 2022-03-13:
@Taoismer, it may not be a MetaDrive problem. 

Instead, you should check your graphics card driver at first by using commands like ```glxinfo -B``` and so on. This can help you figure out if the OpenGL on your machine is currently using software rendering. If so, you may have to install/update your graphics card driver to make sure OpenGL using hardware rendering.

I could not help more but provide you a post https://askubuntu.com/questions/68028/how-do-i-check-if-ubuntu-is-using-my-nvidia-graphics-card, which might be helpful.

Quanyi

closed 2022-04-01 by QuanyiLi