metadriverse/metadrive #155: Error when installing the simulator with Python 3.10: "ERROR: Failed building wheel for numpy"

rondey, 2022-03-13:
Hello, the installation of the simulator in Python 3.10 fails on both Linux and Windows.
I'm facing this problem using the commands "pip3 install -e ." and "pip3 install metadrive-simulator". 
Python 3.8 and 3.9 don't have the same problem.

Here's the error:

'''
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /tmp/pip-standalone-pip-guo5mn91/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-_kr6vvqp/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel scikit-build cmake pip 'numpy==1.13.3; python_version=='"'"'3.6'"'"' and platform_machine != '"'"'aarch64'"'"' and platform_machine != '"'"'arm64'"'"'' 'numpy==1.19.3; python_version>='"'"'3.6'"'"' and sys_platform == '"'"'linux'"'"' and platform_machine == '"'"'aarch64'"'"'' 'numpy==1.21.0; python_version>='"'"'3.6'"'"' and sys_platform == '"'"'darwin'"'"' and platform_machine == '"'"'arm64'"'"'' 'numpy==1.14.5; python_version=='"'"'3.7'"'"' and platform_machine != '"'"'aarch64'"'"' and platform_machine != '"'"'arm64'"'"'' 'numpy==1.17.3; python_version=='"'"'3.8'"'"' and platform_machine != '"'"'aarch64'"'"' and platform_machine != '"'"'arm64'"'"'' 'numpy==1.19.3; python_version>='"'"'3.9'"'"' and platform_machine != '"'"'aarch64'"'"' and platform_machine != '"'"'arm64'"'"''
       cwd: None
  Complete output (971 lines):
  Ignoring numpy: markers 'python_version == "3.6" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
  Ignoring numpy: markers 'python_version >= "3.6" and sys_platform == "linux" and platform_machine == "aarch64"' don't match your environment
  Ignoring numpy: markers 'python_version >= "3.6" and sys_platform == "darwin" and platform_machine == "arm64"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.7" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.8" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
  Collecting setuptools
    Using cached setuptools-60.9.3-py3-none-any.whl (1.1 MB)
  Collecting wheel
    Using cached wheel-0.37.1-py2.py3-none-any.whl (35 kB)
  Collecting scikit-build
    Using cached scikit_build-0.13.1-py2.py3-none-any.whl (75 kB)
  Collecting cmake
    Using cached cmake-3.22.2-py2.py3-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (22.7 MB)
  Collecting pip
    Using cached pip-22.0.4-py3-none-any.whl (2.1 MB)
  Collecting numpy==1.19.3
    Using cached numpy-1.19.3.zip (7.3 MB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
      Preparing wheel metadata: started
      Preparing wheel metadata: finished with status 'done'
  Collecting distro
    Using cached distro-1.7.0-py3-none-any.whl (20 kB)
  Collecting packaging
    Using cached packaging-21.3-py3-none-any.whl (40 kB)
  Collecting pyparsing!=3.0.5,>=2.0.2
    Using cached pyparsing-3.0.7-py3-none-any.whl (98 kB)
  Building wheels for collected packages: numpy
    Building wheel for numpy (PEP 517): started
    Building wheel for numpy (PEP 517): still running...
    Building wheel for numpy (PEP 517): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /tmp/tmp187pkpyv_in_process.py build_wheel /tmp/tmpx6k646is
         cwd: /tmp/pip-install-zir545b5/numpy_a878dba5431048fba439d8ccc2f1d429
    Complete output (930 lines):
    setup.py:67: RuntimeWarning: NumPy 1.19.3 may not yet support Python 3.10.
      warnings.warn(
    Running from numpy source directory.
    /tmp/pip-install-zir545b5/numpy_a878dba5431048fba439d8ccc2f1d429/tools/cythonize.py:67: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
      from distutils.version import LooseVersion
    numpy/random/_bounded_integers.pxd.in has not changed
    Processing numpy/random/_bounded_integers.pyx
    numpy/random/mtrand.pyx has not changed
    numpy/random/_pcg64.pyx has not changed
    numpy/random/bit_generator.pyx has not changed
    numpy/random/_sfc64.pyx has not changed
    numpy/random/_common.pyx has not changed
    numpy/random/_bounded_integers.pyx.in has not changed
    numpy/random/_philox.pyx has not changed
    numpy/random/_generator.pyx has not changed
    numpy/random/_mt19937.pyx has not changed
    Cythonizing sources
    blas_opt_info:
    blas_mkl_info:
    customize UnixCCompiler
      libraries mkl_rt not found in ['/usr/local/lib64', '/usr/local/lib', '/usr/lib64', '/usr/lib', '/usr/lib/']
      NOT AVAILABLE
[... 501 lines cut ...]
    copying numpy/lib/tests/test_twodim_base.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_mixins.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_utils.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_shape_base.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_index_tricks.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/__init__.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_recfunctions.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_arraypad.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_nanfunctions.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_function_base.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_financial.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_ufunclike.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_arrayterator.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    copying numpy/lib/tests/test_histograms.py -> build/lib.linux-x86_64-3.10/numpy/lib/tests
    creating build/lib.linux-x86_64-3.10/numpy/linalg
    copying numpy/linalg/linalg.py -> build/lib.linux-x86_64-3.10/numpy/linalg
    copying numpy/linalg/__init__.py -> build/lib.linux-x86_64-3.10/numpy/linalg
    copying numpy/linalg/setup.py -> build/lib.linux-x86_64-3.10/numpy/linalg
    creating build/lib.linux-x86_64-3.10/numpy/linalg/tests
    copying numpy/linalg/tests/test_linalg.py -> build/lib.linux-x86_64-3.10/numpy/linalg/tests
    copying numpy/linalg/tests/test_deprecations.py -> build/lib.linux-x86_64-3.10/numpy/linalg/tests
    copying numpy/linalg/tests/test_regression.py -> build/lib.linux-x86_64-3.10/numpy/linalg/tests
    copying numpy/linalg/tests/test_build.py -> build/lib.linux-x86_64-3.10/numpy/linalg/tests
    copying numpy/linalg/tests/__init__.py -> build/lib.linux-x86_64-3.10/numpy/linalg/tests
'''

rondey, 2022-03-13:
PART 2/2:

'''
    creating build/lib.linux-x86_64-3.10/numpy/ma
    copying numpy/ma/mrecords.py -> build/lib.linux-x86_64-3.10/numpy/ma
    copying numpy/ma/extras.py -> build/lib.linux-x86_64-3.10/numpy/ma
    copying numpy/ma/__init__.py -> build/lib.linux-x86_64-3.10/numpy/ma
    copying numpy/ma/bench.py -> build/lib.linux-x86_64-3.10/numpy/ma
    copying numpy/ma/testutils.py -> build/lib.linux-x86_64-3.10/numpy/ma
    copying numpy/ma/setup.py -> build/lib.linux-x86_64-3.10/numpy/ma
    copying numpy/ma/timer_comparison.py -> build/lib.linux-x86_64-3.10/numpy/ma
    copying numpy/ma/core.py -> build/lib.linux-x86_64-3.10/numpy/ma
    creating build/lib.linux-x86_64-3.10/numpy/ma/tests
    copying numpy/ma/tests/test_deprecations.py -> build/lib.linux-x86_64-3.10/numpy/ma/tests
    copying numpy/ma/tests/test_regression.py -> build/lib.linux-x86_64-3.10/numpy/ma/tests
    copying numpy/ma/tests/test_mrecords.py -> build/lib.linux-x86_64-3.10/numpy/ma/tests
    copying numpy/ma/tests/test_subclassing.py -> build/lib.linux-x86_64-3.10/numpy/ma/tests
    copying numpy/ma/tests/test_old_ma.py -> build/lib.linux-x86_64-3.10/numpy/ma/tests
    copying numpy/ma/tests/__init__.py -> build/lib.linux-x86_64-3.10/numpy/ma/tests
    copying numpy/ma/tests/test_extras.py -> build/lib.linux-x86_64-3.10/numpy/ma/tests
    copying numpy/ma/tests/test_core.py -> build/lib.linux-x86_64-3.10/numpy/ma/tests
    creating build/lib.linux-x86_64-3.10/numpy/matrixlib
    copying numpy/matrixlib/defmatrix.py -> build/lib.linux-x86_64-3.10/numpy/matrixlib
    copying numpy/matrixlib/__init__.py -> build/lib.linux-x86_64-3.10/numpy/matrixlib
    copying numpy/matrixlib/setup.py -> build/lib.linux-x86_64-3.10/numpy/matrixlib
    creating build/lib.linux-x86_64-3.10/numpy/matrixlib/tests
    copying numpy/matrixlib/tests/test_interaction.py -> build/lib.linux-x86_64-3.10/numpy/matrixlib/tests
    copying numpy/matrixlib/tests/test_regression.py -> build/lib.linux-x86_64-3.10/numpy/matrixlib/tests
    copying numpy/matrixlib/tests/test_matrix_linalg.py -> build/lib.linux-x86_64-3.10/numpy/matrixlib/tests
    copying numpy/matrixlib/tests/test_multiarray.py -> build/lib.linux-x86_64-3.10/numpy/matrixlib/tests
    copying numpy/matrixlib/tests/test_masked_matrix.py -> build/lib.linux-x86_64-3.10/numpy/matrixlib/tests
    copying numpy/matrixlib/tests/test_numeric.py -> build/lib.linux-x86_64-3.10/numpy/matrixlib/tests
    copying numpy/matrixlib/tests/test_defmatrix.py -> build/lib.linux-x86_64-3.10/numpy/matrixlib/tests
    copying numpy/matrixlib/tests/__init__.py -> build/lib.linux-x86_64-3.10/numpy/matrixlib/tests
    creating build/lib.linux-x86_64-3.10/numpy/polynomial
    copying numpy/polynomial/_polybase.py -> build/lib.linux-x86_64-3.10/numpy/polynomial
    copying numpy/polynomial/hermite_e.py -> build/lib.linux-x86_64-3.10/numpy/polynomial
    copying numpy/polynomial/polyutils.py -> build/lib.linux-x86_64-3.10/numpy/polynomial
    copying numpy/polynomial/chebyshev.py -> build/lib.linux-x86_64-3.10/numpy/polynomial
    copying numpy/polynomial/polynomial.py -> build/lib.linux-x86_64-3.10/numpy/polynomial
    copying numpy/polynomial/legendre.py -> build/lib.linux-x86_64-3.10/numpy/polynomial
    copying numpy/polynomial/laguerre.py -> build/lib.linux-x86_64-3.10/numpy/polynomial
    copying numpy/polynomial/__init__.py -> build/lib.linux-x86_64-3.10/numpy/polynomial
    copying numpy/polynomial/setup.py -> build/lib.linux-x86_64-3.10/numpy/polynomial
    copying numpy/polynomial/hermite.py -> build/lib.linux-x86_64-3.10/numpy/polynomial
    creating build/lib.linux-x86_64-3.10/numpy/polynomial/tests
    copying numpy/polynomial/tests/test_polynomial.py -> build/lib.linux-x86_64-3.10/numpy/polynomial/tests
    copying numpy/polynomial/tests/test_laguerre.py -> build/lib.linux-x86_64-3.10/numpy/polynomial/tests
    copying numpy/polynomial/tests/test_polyutils.py -> build/lib.linux-x86_64-3.10/numpy/polynomial/tests
    copying numpy/polynomial/tests/test_hermite_e.py -> build/lib.linux-x86_64-3.10/numpy/polynomial/tests
    copying numpy/polynomial/tests/test_printing.py -> build/lib.linux-x86_64-3.10/numpy/polynomial/tests
    copying numpy/polynomial/tests/test_chebyshev.py -> build/lib.linux-x86_64-3.10/numpy/polynomial/tests
    copying numpy/polynomial/tests/test_hermite.py -> build/lib.linux-x86_64-3.10/numpy/polynomial/tests
    copying numpy/polynomial/tests/__init__.py -> build/lib.linux-x86_64-3.10/numpy/polynomial/tests
    copying numpy/polynomial/tests/test_legendre.py -> build/lib.linux-x86_64-3.10/numpy/polynomial/tests
    copying numpy/polynomial/tests/test_classes.py -> build/lib.linux-x86_64-3.10/numpy/polynomial/tests
    creating build/lib.linux-x86_64-3.10/numpy/random
    copying numpy/random/_pickle.py -> build/lib.linux-x86_64-3.10/numpy/random
    copying numpy/random/__init__.py -> build/lib.linux-x86_64-3.10/numpy/random
    copying numpy/random/setup.py -> build/lib.linux-x86_64-3.10/numpy/random
    creating build/lib.linux-x86_64-3.10/numpy/random/tests
    copying numpy/random/tests/test_random.py -> build/lib.linux-x86_64-3.10/numpy/random/tests
    copying numpy/random/tests/test_direct.py -> build/lib.linux-x86_64-3.10/numpy/random/tests
    copying numpy/random/tests/test_regression.py -> build/lib.linux-x86_64-3.10/numpy/random/tests
    copying numpy/random/tests/test_randomstate_regression.py -> build/lib.linux-x86_64-3.10/numpy/random/tests
[... cut ...] platform == '"'"'darwin'"'"' and platform_machine == '"'"'arm64'"'"'' 'numpy==1.14.5; python_version=='"'"'3.7'"'"' and platform_machine != '"'"'aarch64'"'"' and platform_machine != '"'"'arm64'"'"'' 'numpy==1.17.3; python_version=='"'"'3.8'"'"' and platform_machine != '"'"'aarch64'"'"' and platform_machine != '"'"'arm64'"'"'' 'numpy==1.19.3; python_version>='"'"'3.9'"'"' and platform_machine != '"'"'aarch64'"'"' and platform_machine != '"'"'arm64'"'"'' Check the logs for full command output.
'''

QuanyiLi, 2022-03-13:
Hi, rondey,

Thanks for your feedback. 

We officially recommend using Python3.7, while 3.8 is also ok in some circumstances. This is because the game engine Panda3D lacks the support to Python3.10. 

Personally, I think Python3.7 and 3.8 are enough for a lot of existing projects. So please use Python3.7 instead : )

Quanyi

closed 2022-03-24 by QuanyiLi