metadriverse/metadrive #156: Constant FPS mode

bbenja, 2022-03-14:
Is there a way to set constant fps mode?
I tried `env.engine.force_fps.toggle()`: then `env.engine.force_fps.fps` is showing 50 but visualisation is showing 10-16 fps in the top right corner. Is there any other way? Thanks in advance!

QuanyiLi, 2022-03-15:
Hi,

You could press key ```F``` to switch the mode between real-time-simulation mode and unlimited FPS mode.

Quanyi

bbenja, 2022-03-15:
That's the 'env.engine.force_fps.toggle()' if I'm not mistaken, I tried it, but it's still showing about 10 fps on screen.

pengzhenghao, 2022-03-15:
Oh, why application you are using? For example, in MARL environment, if the number of concurrent vehicles to be as large as 50+ and your compute is not so powerful, then the FPS will drop to very low rate.

The force FPS only limits the maximal FPS. It will only lower FPS to 50FPS and it can't used to increase FPS if your computer can't support higher FPS.

bbenja, 2022-03-15:
I get it that it can only lower fps.
I'm using single agent with `traffic_density` at 0 or 0.05. I'm using it to capture images from the front camera at (1280, 720). All the images are kept in a list and saved to disk at the end of a run so I/O shouldn't be a problem. It's running on Win10, i7-6700, GTX 1070Ti.
So, I'm just interested in making fps constant. Could setting `physics_world_step_size` to 0.1 limit the simulation to 10fps?

QuanyiLi, 2022-03-15:
Hi @bbenja,

I am pretty sure that the low FPS problem results from the performance bottleneck. 

The requested first-view images are in a too-large size (1280x720), which causes severe latency when conveying these images from GPU memory to CPU memory and thus a low fps. Including the game engine of MetaDrive (Panda3D), existing game engines merely do any optimization to the transmission efficiency, since conveying lots of rendered images from GPU to CPU is a trivial demand for the game industry. However, for RL researchers who train visualmotor policies on simulators, it is reasonable to ask for optimizing the transmission latency. 

I suggest 1) turning off the visualization to improve the efficiency or 2) using first-view image buffer in a smaller size, i.e. 210*140, which could achieve almost 90 FPS on GTX 2060.

Quanyi

pengzhenghao, 2022-03-15:
If you want to capture frames and create video in constant FPS, we have another approach:

```
from panda3d.core import PNMImage

img = PNMImage()
env.engine.win.getScreenshot(img)
img.write("frame001.png")
```

This segment can save the image in current pop up windows to a given destination. 

So suppose you are recording videos and want a fixed frame rate, using this segment to generate many pictures and combine them to video might be a good idea!