metadriverse/metadrive #161: Many baselines performance drop introduced in new refactoring, not sure where is the problem

pengzhenghao, 2022-03-20:
We detect performance drop in many baseline RL algorithms. We are checking where is the problem!

pengzhenghao, 2022-03-20:
[image]

a potential bug?

QuanyiLi, 2022-03-25:
## Update:
#### SafeMetaDrive test results:
[image]
The SAC-Lag and SAC still show similar success rate on SafeMetaDrive Environment.

#### MARL results:
Commits merged in 0822-0830 undermine the performance of MARL agent. I am working on it.

pengzhenghao, 2022-03-28:
Finally we diagnosed the reasons:

* **dynamics randomization**: After a commit in 2021-08-18, we introduce dynamics randomization for default vehicle by default. This also impacts MARL environment since we are using default vehicle for all vehicles in the MARL env.

* **bug in termination handling**: We introduce a bug in 2021-08-30, which mistakenly set all vehicles to done when the environmental episode length hits `horizon`. This is not the native logic that (1) the environment stop spawns new vehicles after environmental episode length hits `horizon` and (2) existing vehicles run until die.

The following are our ablation study on the latest MetaDrive (2022-03-28). `vehicle_model=default` means using dynamics randomization, otherwise using the original static dynamics. `disable_auto_done=True` means fixing the bugs in termination handling.

Note: So `vehicle_model: default, disable_auto_done: false` (black) is current latest MetaDrive and `vehicle_model: static_default, disable_auto_done: true` (green) should be the old implementation where the experiment result in our paper generated.

#### Intersection

[image]

#### Parking Lot

[image]

#### Tollgate

[image]

#### Bottleneck

[image]

#### Roundabout

[image]

#### PGMA (Procedural Generated Multi-Agent environment)

[image]

pengzhenghao, 2022-03-28:
Followup:

I suggest to revert these two changes and make them aligned with the version reported in the environment.

Or do you have any advise (like should we keep the dynamics randomization?) after reading the curves? @QuanyiLi

PR #164 (2022-03-28): Fix MARL long lasting bug: dynamic randomization and max step termination handling
## What changes do you make in this PR?

* Please describe why you create this PR

resolve #161  

## Checklist

* [x] I have merged the latest main branch into current branch.
* [x] I have run `bash scripts/format.sh` before merging.
* Please use "squash and merge" mode.
closed 2022-03-28 by pengzhenghao