metadriverse/metadrive #162: Selection of parameter in Rllib training for SAC agent in MetaDriveEnv and SafeMetaDriveEnv

HenryLHH, 2022-03-23:
What is the proper buffer size / batch size / entropy coefficient for SAC to reproduce the results? I find it hard to reproduce results in SafeMetaDriveEnv.
In https://arxiv.org/pdf/2109.12674.pdf, does the reported success rate of SAÇ in Table1 refer to the training success rate (and no collision, i.e. `safe_rl_env=True`)?

QuanyiLi, 2022-03-24:
Hi Henry,

The hyper-parameter could be found in the appendix. 

And the results reported in Table 1 is **test** success rate, which is the average success rate on test maps. Note that in *safe exploration* experiment the number of maps in the training set and test set is 100 and 50 respectively.  

If you feel it is difficult to reproduce the results with your own algorithms, you can actually find our training scripts (based on *ray/rllib*) in https://github.com/decisionforce/EGPO or https://github.com/decisionforce/HACO, where SAC/PPO/SAC-Lag/PPO-Lag are trained on ```SafeMetaDriveEnv```.

Quanyi

pengzhenghao, 2022-03-24:
According to the MetaDrive white paper, the parameters for SAC is:

[image]

As you can see in SAC baseline of EGPO paper: [Code](https://github.com/decisionforce/EGPO/blob/main/training_script/train_sac_baseline.py) most parameters are following [RLLib default](https://docs.ray.io/en/latest/rllib/rllib-algorithms.html#sac). Combining default in RLLib and our specification in our code, the final config of SAC is:

```
train_batch_size = 256,
learning_starts = 10000,
optimization=dict(actor_learning_rate=1e-4, critic_learning_rate=1e-4, entropy_learning_rate=1e-4),
buffer_size=1M
```

Note that the entropy coefficient is update automatically by RLLib and it is not a pre-defined value.

---

The test success rate is recorded in the test environment, which is defined by test environment config as in [here](https://github.com/decisionforce/EGPO/blob/aba65504795e7d5bf652c8bb9ed1f1fb73d84b45/egpo_utils/common.py#L188)

```
evaluation_config = dict(env_config=dict(
    vehicle_config=dict(use_saver=False, overtake_stat=False),
    safe_rl_env=True,
    start_seed=500,
    environment_num=50,
    horizon=1000,
))
```

---

If possible, could you please share the learning curves and your learning code / config? So that we can help to diagnose what's the problem. Thanks!

HenryLHH, 2022-03-24:
Thanks Zhenghao and Quanyi for your timely reply, I follow most of the config of SAC in EGPO and rllib default one. I think the previous issue is mainly due to I put `safe_rl_env=False` during evaluation, which takes collision into consideration in testing time. I retrain the parameter today, think now the result of SAC seems to be quite consistent with the paper.

Also for SAC_Lag, is the penalty learning rate actually the `k_p` for the update on Lagragian multiplier \lambda? I notice the SAC_Lag baseline has different settings of PID controller in HACO and EPGO.

QuanyiLi, 2022-03-25:
Hi @HenryLHH,

In our practice, we find that using sole integral control to update /lambda makes the training unstable. So our SAC-Lag integrates a PID controller to update the Lagrangian multiplier, so the ```k_p```, ``` k_i```, ```k_d``` are 3 hyperparameters to tune the PID controller. 

Since SAC-Lag is not very hyper-parameter-sensitive, the slight hyperparameter difference between HACO version and EGPO version is acceptable.

Quanyi

Quanyi

closed 2022-04-01 by QuanyiLi