metadriverse/metadrive #169: Colab error.

Alexbeast-CN, 2022-03-31:
I tried both 

`!pip install metadrive-simulator`

and 

`!git clone https://github.com/decisionforce/metadrive.git
!cd metadrive
!pip install -e .`

on Colab, then use the following code the set up gym environment:

```
import metadrive  # Import this package to register the environment!
import gym
import numpy as np
import matplotlib.pyplot as plt
from IPython import display as ipythondisplay
from pyvirtualdisplay import Display
display = Display(visible=0, size=(400, 300))
display.start()

env = gym.make("MetaDrive-v0", config=dict(use_render=True))
env.reset()
for i in range(1000):
    obs, reward, done, info = env.step(env.action_space.sample())  # Use random policy
    env.render()
    if done:
        env.reset()
env.close()
```

But there is an error:

```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
/usr/local/lib/python3.7/dist-packages/gym/envs/registration.py in spec(self, path)
    120         try:
--> 121             return self.env_specs[id]
    122         except KeyError:

KeyError: 'MetaDrive-v0'

During handling of the above exception, another exception occurred:

UnregisteredEnv                           Traceback (most recent call last)
3 frames
/usr/local/lib/python3.7/dist-packages/gym/envs/registration.py in spec(self, path)
    129                 raise error.DeprecatedEnv('Env {} not found (valid versions include {})'.format(id, matching_envs))
    130             else:
--> 131                 raise error.UnregisteredEnv('No registered env with id: {}'.format(id))
    132 
    133     def register(self, id, **kwargs):

UnregisteredEnv: No registered env with id: MetaDrive-v0
```

QuanyiLi, 2022-04-01:
Hi, Daoming

We find there are some problems with the environment registration. 

Before we solve this, you can use ```env = metadrive.envs.MetaDriveEnv({"use_render":True})``` to create the environment instead.

Quanyi

closed 2022-06-30 by Alexbeast-CN

pengzhenghao, 2022-09-21:
@QuanyiLi Do we have any update on this issue?