metadriverse/metadrive #196: How to custom a merging zone  and define my own policy?

yuzhouxianzhi, 2022-07-06:
Hi, @QuanyiLi @pengzhenghao .I am deeply inspired by your work.I have read your papers and the documentation of metadrive. I also learned your video on BiliBili. I want to know about custom map,such the merging zone or inramp. I know that Generalization Config and Map Config,such as config["map_config"]["type"] ="block_num", in the documentation of metadrive.  There is an example: map (int or string): User can set a string or int as the key to generate map in an easy way. For example, config["map"] = 3 means generating a map containing 3 blocks, while config["map"] = "SCrRX" means the first block is Straight, and the following blocks are Circular, InRamp, OutRamp and Intersection. 
      But i do not know where define map configuration like the config["map_config"]["type"] ="block_num" or config ["map"] = "SCrRX" in the scrpit to achieve the custom merging zone or inramp? How to add my policy based on stable_baselines3?
      Looking forward your reply. Thank you very much.

       JinchaoHu

pengzhenghao, 2022-07-08:
Hi @yuzhouxianzhi 

I have added a new example to specify map config:

https://github.com/metadriverse/metadrive/blob/main/metadrive/examples/custom_inramp_env.py

Concretely, we have two options to customize map:

1. Use easy map config like:

```python
MetaDriveEnv({"map": "r"})
```

2. Use detailed map config to specify details:

```python
MetaDriveEnv({"map_config": dict(lane_num=1, type="block_sequence", config="r")})
```

Thanks!

closed 2022-07-08 by pengzhenghao

yuzhouxianzhi, 2022-07-08:
Thank you for your answer！