metadriverse/metadrive #207: RGB Camera returns time-buffered grayscale images

EdAlexAguilar, 2022-08-26:
Hi, 
I am running a vanilla MetaDriveEnv with the rgb camera sensor.
```
veh_config = dict(
    image_source="rgb_camera",
    rgb_camera=(IMG_DIM, IMG_DIM))
```

I wanted to see the images the sensor was producing, so was saving a few of them:
I am using: `from PIL import Image`

```
action = np.array([0,0])
obs, reward, done, info = env.step(action)
img = Image.fromarray(np.array(obs['image']*256,np.uint8))
img.save(f"test.jpeg")
```

I noticed that the images all looked grayscale. And upon further inspection I found the following behavior::
Suppose we want (N,N) images, which should be represented as arrays of size (N,N,3).
Step 0: image[:,:,0] = zeros(N,N)  ; image[:,:,1] = zeros(N,N)  ; image[:,:,2] = zeros(N,N)
Step 1: image[:,:,0] = zeros(N,N)  ; image[:,:,1] = zeros(N,N)  ; image[:,:,2] = m1 
Step 2: image[:,:,0] = zeros(N,N)  ; image[:,:,1] = m1  ; image[:,:,2] = m2
Step 3: image[:,:,0] = m1  ; image[:,:,1] = m2  ; image[:,:,2] = m3

where m1, m2, m3 are (N,N) matrices.

So, the images are in reality displaying 3 different timesteps with the color channels taking the time info (R=t-2,  G=t-1, B=t) . That is why the images look mostly gray, since the values are identical almost everywhere – except where we expect some movement (contours) where we see that lines look colorful and strange.

Apologies if this is expected behavior, and I just had some configuration incorrect.

[image]

QuanyiLi, 2022-08-30:
Hi, 

Sorry for the late reply :). You are correct! It is the expected behavior. 

The RGB mode returns **grayscale image** by default and thus the RGB channel now is the temporal channel. Considering that the RGB image could be converted into a greyscale image without loss of quality, we directly retrieve the greyscale image from the camera buffer and stack these one-channel images in consecutive timesteps so that the CNN can process them easily. 

If the RGB image is necessary for your task, I would be happy to construct the RGB observer for you. But personally, I suggest using a depth camera for a locomotion task.  The depth image is a natural and effective representation of locomotion tasks and is verified in several previous works. 

Quanyi

EdAlexAguilar, 2022-08-30:
Hi!
Many thanks for the reply!
I understand that grayscale + temporal stacking is a very common approach for RL, which is why I suspected this was the expected behavior. In our case, however, we do need the actual RGB image. So if this is something that you can offer, we would be extremely grateful!

As a side note, if the grayscale approach is the intended behavior, then it might be best to update the documentation to clarify that. Since currently the [observation documentation](https://metadrive-simulator.readthedocs.io/en/latest/observation.html) shows actual RGB images there.

Many thanks in advance for looking into it!
-Edgar

QuanyiLi, 2022-08-31:
Hi Edgar,

I created a new branch #208 which enables this new feature. In the new ```rgb-cam``` branch, the returned image will be RGB image by default. By setting the ```vehicle_config["rgb_to_grayscale"] to True```, the RGB image can still be converted to grayscale. 

I have already tested this new implementation with several scripts located at: ```metadrive/tests/vis_functionality``` including ```vis_rgb_cam.py / vis_grayscale.py / profile_rgb_cam.py and vis_shared_camera.py```.  You could simply modify these scripts for testing if this is a bug-free implementation for your purpose. I will merge it into the ```main``` branch after receiving your feedback.

Look forward to your reply : )

Quanyi

closed 2022-09-13 by QuanyiLi