metadriverse/metadrive #209: Running Metadirive on Windows 10 has a problem

benicioolee, 2022-09-10:
The following error occurs when I run Metadrive with Windows:

(metadrive) PS D:\metadrive-main\metadrive\examples> python profile_metadrive.py
Traceback (most recent call last):
  File "profile_metadrive.py", line 5, in <module>
    from metadrive import MetaDriveEnv
  File "d:\metadrive-main\metadrive\__init__.py", line 1, in <module>
    import metadrive.register
  File "d:\metadrive-main\metadrive\register.py", line 73, in <module>
    if env_name not in registry.env_specs:
AttributeError: 'dict' object has no attribute 'env_specs'

What is the reason for the error? metadrive is not compatible with Windows and cannot be run on windows.

QuanyiLi, 2022-09-10:
Hi,

We are sure that MetaDrive is compatible with Windows/MacOS/Linux. 

It seems that your ```gym``` is in an incorrect version so the ```registry``` imported from the ```gym``` is wrong. I tested MetaDrive on my Windows machine with ```gym==0.19.0``` and it works well.

Could you provide us with more information? For example, how do you install MetaDrive and the requirements?

Thanks,
Quanyi

benicioolee, 2022-09-11:
Hi,
Thanks for your help. I have solved the problem by modifying setup.py. Set the version of the gym to 0.19.0 in setup.py. 
Thanks，
lee

closed 2022-09-11 by QuanyiLi