metadriverse/metadrive #211: about panda3d

benicioolee, 2022-09-29:
when i run "python -m metadrive.examples.drive_in_safe_metadrive_env", set use_render=true
the output:
Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
Known pipe types:
  glxGraphicsPipe
(1 aux display modules not yet loaded.)

QuanyiLi, 2022-09-29:
Hi

This is the normal output. 

Are there any problems with this script, like invalid keyboard input or no window popping up? 

Quanyi

benicioolee, 2022-09-29:
no window popping up

QuanyiLi, 2022-09-30:
That is weird. Could you provide more information about your platform, like your OS, and GPU?

benicioolee, 2022-10-02:
After I installed carla, metadrive wouldn't pop up the window, I have solved the problem by reinstalling the following packages.
    panda3d~=1.10.8",
    "panda3d-gltf",
    "panda3d-simplepbr",
    "pillow",
    "pytest",
    "opencv-python-headless",

closed 2022-10-04 by QuanyiLi