metadriverse/metadrive #233: I encountered an error at an unknown location during runtime. Hello

shushushulian, 2022-11-06:
Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
**Known pipe types:
  wglGraphicsPipe
(all display modules loaded.)**

pengzhenghao, 2022-11-06:
Where is the error?

shushushulian, 2022-11-06:
When running the example folder drive_in_single_agent_env.py
I figured it was engine_utils.py——**cls.singleton**
**def initialize_engine(env_global_config):
    cls = BaseEngine
    # print(env_global_config)
    if cls.singleton is None:
        # assert cls.global_config is not None, "Set global config before initialization BaseEngine"
        cls.singleton = cls(env_global_config)
    else:
        raise PermissionError("There should be only one BaseEngine instance in one process")
    return cls.singleton**

I reinstalled the panda3D

pengzhenghao, 2022-11-06:
I don’t think you need to reinstall panda. You need to restart the jupyternotebook or the process where you once initialized the MetaDrive environment.

here is the story: you can only create one MetaDrive environment in one process. Unless, you explicitly call the close function of the first environment instance.

env1 = MetaDriveEnv()
env1.close()
env2 = MetaDriveEnv()

shushushulian, 2022-11-07:
Thank you very much.

closed 2022-11-07 by pengzhenghao