metadriverse/metadrive #249: BREAKING NEWS: panda3d-1.10.12 makes MetaDrive works in Mac with M1 chip!!

pengzhenghao, 2022-12-02:
I think it's a good time to consider upgrade the dependency of panda3d!!!!!!!

pengzhenghao, 2022-12-02:
Report:

Can run profile_metadrive example,

but can't use the Panda3d native renderer as:

```
pengzhenghao@ examples % python drive_in_multi_agent_env.py
Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
Known pipe types:
  CocoaGraphicsPipe
(all display modules loaded.)
:display:cocoadisplay(error): Could not find a usable pixel format.
:ShowBase(warning): Unable to open 'onscreen' window.
```

QuanyiLi, 2022-12-02:
Do they officially announce that it supports M1 chips? If so, you can report this bug to the panda3D forum and ask for help

pengzhenghao, 2023-01-09:
The Panda3d rendering is still problematic. But I can run successfully without Panda3d rendering and I can workaround by using Pygame renderer!! 

So it helps a lot. I can debug in my mac!

pengzhenghao, 2023-02-28:
We have updated Panda3d requirement to allow high version that are compatible with M1/2 Mac

closed 2023-02-28 by pengzhenghao