metadriverse/metadrive #252: how to add user-specific network into metadrive

SHITIANYU-hue, 2022-12-07:
Hi, I am wondering how to add a user-specific network to the simulator, such as Toronto's road network downloaded from the open street map. thanks!

QuanyiLi, 2022-12-07:
Hi Tianyu,

Thanks for your interest, while loading OpenDriveMap or OpenStreetMap into the simulator is **NOT** fully supported currently. 

In ```metadrive/tests/vis_opendrive_map/vis_carla_town_1.py```, we tested loading CARLA town 1, which is in OpenDrive format. If running this script, you will find that the road network is not smooth enough, and it still needs more time for us to improve the APIs. 

But you could still try loading your map data with this script after converting the ```OpenStreetMap``` to ```OpenDriveMap``` format. If your data is compatible with the existing data-loading workflow of MetaDrive, I would be very happy to finish this feature as soon as possible, so that you can use MetaDrive for your research. 

Look forward to hearing from you!

Quanyi

SHITIANYU-hue, 2022-12-23:
Hi Quanyi

Thanks! I can successfully load the environment's network. I am wondering if it can support training in a real-world scenario.

[image]

QuanyiLi, 2022-12-25:
For training in the real world, we recommend using Waymo-motion dataset: https://github.com/metadriverse/metadrive-scenario

Each case in this data is collected in the real world with real road network and real traffics. But for loading real maps, like the Toronto map, MetaDrive is not fully supported yet.

closed 2023-02-11 by QuanyiLi