metadriverse/metadrive #261: copy.copy() should always be replaced by copy.deepcopy()

pengzhenghao, 2022-12-22:
As title.

closed 2023-02-07 by pengzhenghao

QuanyiLi, 2023-02-24:
For config and some ```class Config```objects, it works. But for other objects like ```class Lane```, we don't like it to work in that way, as there might be some Panda/Bullet objects in the instance variables which can not be deep copied. In this case, we actually want a shallow copy. 

If you run the main branch with ```debug=True```, you will see a lot of error messages saying bullet objects can not be deep copied. I already fixed it in the #310

pengzhenghao, 2023-02-24:
Forget this issue! The copy problem is largely resolved when I fixed the memory leak. And everything here now are just quite good and don’t need to changeBest,Zhenghao