metadriverse/metadrive #262: very slow fps

SHITIANYU-hue, 2022-12-22:
I run it on my computer, but the FPS is just 0.x,  if it is possible to increase the FPS rate?

[image]

QuanyiLi, 2022-12-22:
Hi, Tianyu

The reported FPS of on-screen mode depends on your hardware and the complexity of the scenario. You can simply test the FPS in a single-agent environment. If the FPS in the single-agent environment is in the normal range (30~60), the FPS drop in this MARL environment must be caused by the complex MARL environment. In this case, you should update your hardware, especially the graphics card. Actually, from the broken skybox in the image above, I guess that there is no discrete graphics card on your machine. This will make it difficult to render the complex MARL scene. 

Besides, this FPS in your figure is the on-screen mode FPS, while running in off-screen mode will greatly increase the FPS since rendering is disabled. In this case, the lack of a discrete graphics card will not affect your FPS. You can test the FPS in off-screen mode by running ```examples/profile_metadrive.py```, which will report the off-screen FPS in the single-agent environment. If it is still ```0.X```, please report it to us and we would be happy to help you with it.

Quanyi

pengzhenghao, 2022-12-23:
Run `metadrive/examples/profile_metadrive.py`, I can reach astonishing 1300 FPS:

[image]

pengzhenghao, 2022-12-23:
In Multi-agent Roundabout environment, I can still reach 110 FPS (environmental steps per second).

Consider we have 40 vehicles there, the total number of samples you can get in each second is almost ~4000 Transitions. I think we can't say the FPS is "very slow" according to this data.

[image]

pengzhenghao, 2022-12-23:
You can try this file: 

https://github.com/metadriverse/metadrive/blob/a63faa909c95eedb1283f1f42070c28da79c7c6e/metadrive/examples/profile_metadrive_marl.py

to see if the result aligns

SHITIANYU-hue, 2022-12-23:
Yes, i tried to run examples/profile_metadrive.py.  And it is very fast, if it is possible to increase speed in rendering mode, if I want to see visualization results?

[image]

SHITIANYU-hue, 2022-12-23:
However, when I use render mode, the FPS is still very slow; I wonder if it is possible to support recording off-screen. So, I could see how the agent drive using the learned policy. 

[image]

QuanyiLi, 2022-12-23:
@SHITIANYU-hue Yes, you can get the screenshot for each frame and combine them to get a video.
The following code is for saving the rendered image into a PNG file, then you can use ```ffmpeg``` to combine them.
```
folder = "your/dir/path"
for frame in range(1000):
    ...
    env.step(action)
    ...
    img = PNMImage()
    env.engine.win.getScreenshot(img)
    img.write("{}_{}.png".format(folder,frame))
```

QuanyiLi, 2022-12-23:
Also, if it still costs too much time for offscreen rendering. You can try use ```top-down render```, which is built on ```pygame``` and thus would be more efficient.

closed 2023-02-07 by pengzhenghao