metadriverse/metadrive #266: Errors when running metadrive.tests.scripts.generate_video_for_image_obs

YouSonicAI, 2022-12-24:
In metadrive directory,  I ran python -m metadrive.tests.scripts.generate_video_for_image_obs , then it reported an error as below:
python -m metadrive.tests.scripts.generate_video_for_image_obs

> Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
> :display(warning): Unable to load libpandagles2.so: No error.
> Known pipe types:
> (all display modules loaded.)
> Traceback (most recent call last):
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/runpy.py", line 193, in _run_module_as_main
>     "__main__", mod_spec)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/runpy.py", line 85, in _run_code
>     exec(code, run_globals)
>   File "/Users/queenie/Documents/metadrive/metadrive/tests/scripts/generate_video_for_image_obs.py", line 157, in <module>
>     env.reset()
>   File "/Users/queenie/Documents/metadrive/metadrive/envs/base_env.py", line 333, in reset
>     self.lazy_init()  # it only works the first time when reset() is called to avoid the error when render
>   File "/Users/queenie/Documents/metadrive/metadrive/envs/base_env.py", line 234, in lazy_init
>     self.engine = initialize_engine(self.config)
>   File "/Users/queenie/Documents/metadrive/metadrive/engine/engine_utils.py", line 11, in initialize_engine
>     cls.singleton = cls(env_global_config)
>   File "/Users/queenie/Documents/metadrive/metadrive/engine/base_engine.py", line 28, in __init__
>     EngineCore.__init__(self, global_config)
>   File "/Users/queenie/Documents/metadrive/metadrive/engine/core/engine_core.py", line 135, in __init__
>     super(EngineCore, self).__init__(windowType=self.mode)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 339, in __init__
>     self.openDefaultWindow(startDirect = False, props=props)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 1021, in openDefaultWindow
>     self.openMainWindow(*args, **kw)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 1056, in openMainWindow
>     self.openWindow(*args, **kw)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 766, in openWindow
>     win = func()
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 752, in <lambda>
>     callbackWindowDict = callbackWindowDict)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 818, in _doOpenWindow
>     self.makeDefaultPipe()
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 648, in makeDefaultPipe
>     "No graphics pipe is available!\n"
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/directnotify/Notifier.py", line 130, in error
>     raise exception(errorString)
> Exception: No graphics pipe is available!
> Your Config.prc file must name at least one valid panda display
> library via load-display or aux-display.
> (drivemeta) ➜  metadrive git:(main) python -m metadrive.tests.scripts.generate_video_for_image_obs
> Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
> :display(warning): Unable to load libpandagles2.so: No error.
> Known pipe types:
> (all display modules loaded.)
> Traceback (most recent call last):
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/runpy.py", line 193, in _run_module_as_main
>     "__main__", mod_spec)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/runpy.py", line 85, in _run_code
>     exec(code, run_globals)
>   File "/Users/queenie/Documents/metadrive/metadrive/tests/scripts/generate_video_for_image_obs.py", line 157, in <module>
>     env.reset()
>   File "/Users/queenie/Documents/metadrive/metadrive/envs/base_env.py", line 333, in reset
>     self.lazy_init()  # it only works the first time when reset() is called to avoid the error when render
>   File "/Users/queenie/Documents/metadrive/metadrive/envs/base_env.py", line 234, in lazy_init
>     self.engine = initialize_engine(self.config)
>   File "/Users/queenie/Documents/metadrive/metadrive/engine/engine_utils.py", line 11, in initialize_engine
>     cls.singleton = cls(env_global_config)
>   File "/Users/queenie/Documents/metadrive/metadrive/engine/base_engine.py", line 28, in __init__
>     EngineCore.__init__(self, global_config)
>   File "/Users/queenie/Documents/metadrive/metadrive/engine/core/engine_core.py", line 135, in __init__
>     super(EngineCore, self).__init__(windowType=self.mode)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 339, in __init__
>     self.openDefaultWindow(startDirect = False, props=props)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 1021, in openDefaultWindow
>     self.openMainWindow(*args, **kw)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 1056, in openMainWindow
>     self.openWindow(*args, **kw)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 766, in openWindow
>     win = func()
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 752, in <lambda>
>     callbackWindowDict = callbackWindowDict)
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 818, in _doOpenWindow
>     self.makeDefaultPipe()
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/showbase/ShowBase.py", line 648, in makeDefaultPipe
>     "No graphics pipe is available!\n"
>   File "/Users/queenie/anaconda3/envs/drivemeta/lib/python3.7/site-packages/direct/directnotify/Notifier.py", line 130, in error
>     raise exception(errorString)
> Exception: No graphics pipe is available!
> Your Config.prc file must name at least one valid panda display
> library via load-display or aux-display.

SHITIANYU-hue, 2022-12-24:
Yes, i have the same issue

QuanyiLi, 2022-12-25:
@SHITIANYU-hue @YouSonicAI 
Unfortunately, this script is broken. But you can use the following script to generate videos from stored images.
```
def image_to_video(seed):
    image_folder = '{}'.format(seed)
    video_name = '{}.mp4'.format(seed)

    images = [img for img in os.listdir(image_folder) if img.endswith(".png")]
    images.sort(key=lambda x: int(x[:-4]))
    frame = cv2.imread(os.path.join(image_folder, images[0]))
    height, width, layers = frame.shape

    video = cv2.VideoWriter(video_name, 0, 30, (width, height))

    for image in images:
        video.write(cv2.imread(os.path.join(image_folder, image)))

    cv2.destroyAllWindows()
    video.release()
```

This script reads the images in ```seed``` folder and sorts them according to their name. After that, ```cv2
``` is used to generate videos named ```seed.mp4```.

This code is tested at https://github.com/metadriverse/metadrive-scenario/blob/main/metadrive_scenario/utils/record_scenario.py, where you can get some useful snippets

closed 2023-02-11 by QuanyiLi