metadriverse/metadrive #270: Deterministic runs

sguysc, 2023-01-12:
Hello,
I'm trying to reproduce runs and it works fine when there is no other traffic but when traffic is present the runs are not deterministic. Before Here's the code for reproducibility:
```python3
import random
import numpy as np
import matplotlib.pyplot as plt
from metadrive import MetaDriveEnv 
from metadrive.examples import expert

rnd_seed = 123

RENDER = True

config = dict(
        use_render=False,
        manual_control=False,
        traffic_density=0.1, # or 0 to be deterministic
        traffic_mode="respawn",
        environment_num=1,
        random_agent_model=False,
        random_lane_width=False,
        random_lane_num=False,
        random_traffic=False,
        start_seed=0,
        force_destroy=True,
        map="SOT",   
    )

env = MetaDriveEnv(config)
print("Starting the environment ...\n")

for episode in range(3):
    random.seed(rnd_seed)
    np.random.seed(rnd_seed)
    obs = env.reset(force_seed=0)

    obs_vec = []
    ep_crashes, ep_cost = 0.0, 0.0
    frames = []
    for t in range(1000):
        action = expert(env.vehicle, deterministic=True)
        obs, reward, done, info = env.step(action)
        obs_vec.append(obs)
        if(RENDER):
            frame = env.render(mode="top_down", track_target_vehicle=True)
            frames.append(frame)
        ep_crashes += info['crash']*1.0
        ep_cost += info['cost']
        if info['arrive_dest']:
            print(f'done after {t} steps')
            break

    all_data = np.array(obs_vec)
    print("The returned reward:\n {}.\n".format(info['episode_reward']))
    print("crashes:\n {}\n".format(ep_crashes))
    print("The returned information:\n {}.".format(info))

    plt.plot(all_data[:,0], label=f'episode {episode}')

env.close()
```
So as [you can see](https://ibb.co/JFcNrRN), I'm comparing the first lidar ray throughout the simulation for 3 runs and they are not the same. What's more interesting, if you look at the figure or run the code, is that every odd iteration is the same, and every even iteration is the same (at least in the lidar ray).

I disabled `random_traffic`, the expert is `deterministic`, the `seed` is the same as well as setting the seed in `random` and `numpy` (I guess it wouldn't hurt), and setting a specific `map`.

What else could I be missing? Or ideas for a fix?

QuanyiLi, 2023-01-13:
Hi, 

Thanks for your feedback. 

I run the code on the latest ```main``` branch and cannot reproduce it. Three runs have the same reward and episode length. Also, I add a simple asset for comparing the observation between three episodes:
```
    epi_len = len(obs_vec)
    assert len(episode_obs[0]) == len(episode_obs[1]) == len(episode_obs[2])
    for i in range(epi_len):
        assert all(episode_obs[0][i] == episode_obs[1][i]) and all(episode_obs[1][i] == episode_obs[2][i])
```
No errors are thrown. 

Did you run on the latest main branch? Recently, I merged a new modification related to the reproducing problem into the ```main```. You might miss it if you didn't update your local ```main``` branch. 

Quanyi

sguysc, 2023-01-16:
Thanks for coming back to me!
I am running the latest `main`, I've only downloaded metadrive a few days ago. The only difference I would say,
is that I have Panda3D==1.10.13.
It's very consistent for me. If I record just the expert's commands on the first iteration and run them in the second 
iteration, then I get the exact same runs. But if not, at some point they diverge. I've also tried to "fool it" and 
reset it twice or just run a few steps, but that won't work, seems like I need to run until the end of the episode.

QuanyiLi, 2023-01-20:
Using a string like ```SOT``` is enough I think. It is a shortcut for generating maps. The full configuration should be like:
```
map_config={
        BaseMap.GENERATE_TYPE: MapGenerateMethod.BLOCK_SEQUENCE,
        BaseMap.GENERATE_CONFIG: "SOT",
        BaseMap.LANE_WIDTH: 3.5,
        BaseMap.LANE_NUM: 3,
        "exit_length": 50,
    }

config = dict(
        use_render=False,
        manual_control=False,
        traffic_density=0.1, #10, #0.15,
        traffic_mode="respawn",
        environment_num=1,
        random_agent_model=False,
        random_lane_width=False,
        random_lane_num=False,
        random_traffic=False,
        start_seed=0,
        force_destroy=True,
        map_config=map_config,
        random_spawn_lane_index=False,
        success_reward=100.,
    )
env = MetaDriveEnv(config)

```

closed 2023-04-23 by QuanyiLi