metadriverse/metadrive #271: Problem with visualisation

mainakpal08, 2023-01-17:
`python -m metadrive.examples.drive_in_single_agent_env `

produces

```
Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
Known pipe types:
  CocoaGraphicsPipe
(all display modules loaded.)
:display:cocoadisplay(error): Could not find a usable pixel format.
:ShowBase(warning): Unable to open 'onscreen' window.
```

Device: MacOS 13.1 M1 arm64 (`macOS-13.1-arm64-arm-64bit`)
Python: `3.9.15`

QuanyiLi, 2023-01-17:
According to [this issue](https://github.com/metadriverse/metadrive/issues/249), maybe you can update panda to 1.10.12 and use the Pygame renderer for using MetaDrive on MAC with new chips.

As @pengzhenghao has more experience with MAC+MetaDrive, I will also call for his help. 

@pengzhenghao Could you help with this MAC problem?

mainakpal08, 2023-01-17:
Thanks for the reply @QuanyiLi !

@pengzhenghao , could you please help me how to use the Pygame renderer ?

pengzhenghao, 2023-01-17:
Mac is not supported yet for the Panda3d native renderer. You can simply using Pygame renderer instead!

pengzhenghao, 2023-01-17:
[image]

By changing the script with the above modifications, you can see a pop up window showing the pygame renderer. Note that `python -m metadrive.examples.drive_in_single_agent_env ` this script is used for demo so you might see that the cars are not moving.

In short, it is very easy to use pygame renderer, just add one line to your code: `env.render("topdown")`

mainakpal08, 2023-01-17:
@pengzhenghao thanks! It works perfectly!

Also setting `manual_control` to `True` makes the car moving too!

pengzhenghao, 2023-01-17:
Great!

closed 2023-01-17 by pengzhenghao