metadriverse/metadrive #275: .profile_metadrive fails for numpy?

smarianimore, 2023-01-30:
Hi everybody, 
I'm trying to run the basic examples of metadrive, but before doing so I run the tests script and it fails as follows:

```
Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
Start to profile the efficiency of MetaDrive with 1000 maps and ~4 vehicles!
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/examples/profile_metadrive.py", line 21, in <module>
    obs = env.reset()
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/envs/base_env.py", line 356, in reset
    self.engine.reset()
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/engine/base_engine.py", line 251, in reset
    manager.reset()
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/manager/map_manager.py", line 55, in reset
    map = self.spawn_object(PGMap, map_config=map_config, random_seed=None)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/manager/map_manager.py", line 27, in spawn_object
    map = self.engine.spawn_object(object_class, auto_fill_random_seed=False, force_spawn=True, *args, **kwargs)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/engine/base_engine.py", line 122, in spawn_object
    obj = object_class(**kwargs)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/map/base_map.py", line 52, in __init__
    self._generate()
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/map/pg_map.py", line 51, in _generate
    self._big_generate(parent_node_path, physics_world)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/map/pg_map.py", line 73, in _big_generate
    big_map.generate(self._config[self.GENERATE_TYPE], self._config[self.GENERATE_CONFIG])
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/algorithm/BIG.py", line 80, in generate
    if self.big_helper_func():
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/algorithm/BIG.py", line 88, in big_helper_func
    self._forward()
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/algorithm/BIG.py", line 132, in _forward
    block = self.sample_block()
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/algorithm/BIG.py", line 111, in sample_block
    block = block_type(
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/pgblock/roundabout.py", line 24, in __init__
    super(Roundabout, self).__init__(*args, **kwargs)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/pgblock/pg_block.py", line 79, in __init__
    super(PGBlock, self).__init__(
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/block/base_block.py", line 35, in __init__
    super(BaseBlock, self).__init__(str(block_index) + self.ID, random_seed, escape_random_seed_assertion=True)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/base_class/base_object.py", line 112, in __init__
    super(BaseObject, self).__init__(name, random_seed, config)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/base_class/base_runnable.py", line 26, in __init__
    self.sample_parameters()
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/base_class/base_runnable.py", line 93, in sample_parameters
    self.update_config(ret)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/base_class/configurable.py", line 28, in update_config
    self._config.update(config, allow_add_new_key=allow_add_new_key)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/utils/config.py", line 141, in update
    self._update_single_item(k, v, allow_add_new_key)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/utils/config.py", line 163, in _update_single_item
    self._set_item(k, v, allow_overwrite)
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/utils/config.py", line 215, in _set_item
    if isinstance(value, (np.float32, np.float64, np.float)):
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive-venv/lib/python3.9/site-packages/numpy/__init__.py", line 284, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'float'
Exception ignored in: <function BaseBlock.__del__ at 0x7fa001773c10>
Traceback (most recent call last):
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/block/base_block.py", line 297, in __del__
    self.destroy()
  File "/Users/sm/PycharmProjects/RomoliGiacomo/metadrive/metadrive/component/block/base_block.py", line 285, in destroy
    if self.block_network is not None:
AttributeError: 'Roundabout' object has no attribute 'block_network'
```

Any suggestion?

QuanyiLi, 2023-01-30:
Hi,
Thanks for your feedback.

I guess you have the wrong ```numpy``` version. Please use ```pip install numpy==1.21``` to reinstall it.

Quanyi

pengzhenghao, 2023-01-30:
> many thanks, now it works

Hi Quanyi, numpy just deprecated np.int np.float and some other useful types. We should probably remove all these types and use builtin python types instead.

QuanyiLi, 2023-01-30:
> Hi Quanyi, numpy just deprecated np.int np.float and some other useful types. We should probably remove all these types and use builtin python types instead.
> […](#)

agree, I will fix it

PR #277 (2023-02-07): Replace deprecated numpy types by builtin types
## What changes do you make in this PR?

* Please describe why you create this PR

close #275

## Checklist

* [ ] I have merged the latest main branch into current branch.
* [ ] I have run `bash scripts/format.sh` before merging.
* Please use "squash and merge" mode.
closed 2023-02-07 by pengzhenghao