metadriverse/metadrive #276: Question about the dead vehicles

liuqianqian-x, 2023-02-07:
Hi authors,

Thanks for providing the open source driving simulator.

When I use a multi-agent environment, the size of information(e.g. observations) obtained by one step is variable because some vehicles are terminated at the exact step. How should this situation be resolved? It seems that it should include the dead vehicles in env.vehicles at the step it dead.

Thanks！

pengzhenghao, 2023-02-07:
Could you provide sample code of your use case? I remember the observations are always aligned with the observation space and has a fixed length. 

Even though you set `config["vehicle_config] = {"lidar": {"num_others": 4}}` you will still get a fixed length observation. (The above config means the observation will include the states of 4-nearest vehicles. If no enough vehicles exist in the neighborhood then all zeros will be filled into the observation for the corresponding dimensions)

> It seems that it should include the dead vehicles in env.vehicles at the step it dead.

For this use case, yes we have! Please use this:

```
vehicles = env.vehicles_including_just_terminated()
```

Then you will have a list of vehicles including those just dead at current step.

liuqianqian-x, 2023-02-08:
> Could you provide sample code of your use case? I remember the observations are always aligned with the observation space and has a fixed length.
>
> Even though you set `config["vehicle_config] = [...]

For example, we have 12 vehicles，after some steps (as shown below), the number of cars will decrease, and then observation will decrease(specifically, the length of `o` will be less), 

```
o, r, d, info = env.step(actions)
```

but sometimes the controllable vehicles will respawn. So how to deal with the changing observation size in the algorithm？

closed 2023-04-23 by QuanyiLi