metadriverse/metadrive #289: blank grey screen

MoreTore, 2023-02-13:
installed as a submodule in exsisting repo. 
ran pip install -e .

when i run the examples I get a blank grey window.

QuanyiLi, 2023-02-14:
Hi,

The versioin installed via pip stays behind the newest ```main``` branch. Thus, it might be some new patches/updates in the latest MetaDrive which cause your problem. 

Could you help me do a simple validation? It is really easy. Just switch to the ```0.2.6``` release branch in your local repo via ```git checkout git checkout releases/0.2.6.0``` and see if the problem is solved or not. If so, my guess is correct. Otherwise, the issue will be complex.

Also, could you provide more information about your platform, *i.e.* OS/CPU/GPU and so on? Maybe I can try reproduce it. Thanks!

Quanyi

MoreTore, 2023-02-15:
Yea I tried releases/0.2.6.0 and it's the same issue. I am going to assume it's my WSL2 setup since it usually has weird issues like this.

QuanyiLi, 2023-02-15:
WSL2? That is interesting. Do WSL2 support opening window from any application now? I remember that it was only optimized to visualize rare applications like Chrome. Or do you use an XServer to do visualization? I am done with dual-boot, if the visualization is supported, I will delete the Ubuntu on my laptop right now, lol

MoreTore, 2023-02-15:
Oh yea I got everything I use running in wsl2 now. There are just different bugs to work out than when running natively. Here is a video of what I'm working on. https://www.youtube.com/watch?v=P-ti8c1esiY&ab_channel=MoreTorque

pengzhenghao, 2023-02-16:
The video is so cool!

closed 2023-02-17 by QuanyiLi