metadriverse/metadrive #290: rgb_camera is slow to render

MoreTore, 2023-02-15:
the fps slows down too much when increasing rgb_camera size. I am trying to send the rgb_camera to another program so that it can run the vision model and send real-time controls back to meadrive. Is there a better way to do this?

import zmq
import lz4.frame as lz4
import argparse
import random
from metadrive import MetaDriveEnv
from metadrive.constants import HELP_MESSAGE
import threading
import numpy as np
import cv2

def main_thread():
    config = dict(
        # controller="joystick",
        use_render=True,
        offscreen_render=True,
        manual_control=True,
        traffic_density=0.0,
        environment_num=100,
        random_agent_model=True,
        random_lane_width=True,
        random_lane_num=True,
        vehicle_config = dict(image_source="rgb_camera", 
                              rgb_camera= (1920,1200),
                              stack_size=1),
        map=4,  # seven block
        start_seed=random.randint(0, 1000),
        window_size=(W, H)
    )
    env = MetaDriveEnv(config)
    try:
        o = env.reset()
        print(HELP_MESSAGE)
        env.vehicle.expert_takeover = False
        context = zmq.Context()
        socket = context.socket(zmq.PUSH)
        socket.bind("tcp://127.0.0.1:5555")
        socket.setsockopt(zmq.SNDHWM, 1)
        socket.setsockopt(zmq.RCVHWM, 1)
        vehicle_socket = context.socket(zmq.PUSH)
        vehicle_socket.bind("tcp://127.0.0.1:5556") # vehicle state
        vehicle_socket.setsockopt(zmq.SNDHWM, 1)
        vehicle_socket.setsockopt(zmq.RCVHWM, 1)
        
        
        assert isinstance(o, dict)
        while True:
            #sm.update(0)
            #throttle_brake_op = sm['carControl'].actuators.accel
            #steer_op = sm['carControl'].actuators.steeringAngleDeg
            o, rewards, dones, step_infos = env.step([0,0))
            try:
                img = o['image'] # (84,84,3,3)
                compressed_image = lz4.compress(img)             
                socket.send(compressed_image, zmq.NOBLOCK)
                vehicle_socket.send_pyobj(step_infos, zmq.NOBLOCK)
            except zmq.error.Again:
                print("Dropped frame")
                pass
            try:
                vehicle_socket.send_pyobj(step_infos, zmq.NOBLOCK)
            except zmq.error.Again:
                print("Dropped vehicle state")
                pass
    except Exception as e:
        raise e
    finally:
        env.close()
        socket.close()
        context.term()
        

if __name__ == "__main__":
    main_thread()

QuanyiLi, 2023-02-15:
No actually. You are right, the FPS decreases as the image size increases, I suggest using a low-resolution image to speed up your program. But I believe that the ceiling of performance is around 100 FPS under 120*80 resolution. The reason is that the game engine has to move the image rendered in GPU to RAM, and then your vision model moves it to GPU again. These actions cost a huge amount of time. 

It is a common issue not only for MetaDrive but for most simulators, like CARLA. As far as I know, only IsaacGym supports reading the rendered images from the GPU memory directly, and the efficiency is not affected by the image size.

QuanyiLi, 2023-02-15:
Your video looks cool. I guess it is lane line detection, right? If so, you could adjust the camera pose to only capture the lane surface, so that you can reduce the size of the image and meanwhile retain as much useful information as possible.

MoreTore, 2023-02-16:
Thanks. It's actually commaai's openpilot. I am looking to for a light-weight simulator to test openpilot on PC. I think I can make metadrive work I just need to improve performance. 

Is there a way to disable rendering for the chase camera and mini map? Headless mode will still render it if I understand correctly. I only need the one rgb_camera observation.

pengzhenghao, 2023-02-16:
> the chase camera and mini map? Headless mode will still render it if I understand correctly. I only need the one rgb_camera observation.

Wow if that's True I think this is a bug. We don't need these two cameras in headless mode.

QuanyiLi, 2023-02-16:
@MoreTore Chase camera and other cameras will still render in headless or offscreen mode. I think @pengzhenghao already creates a new PR to turn off the main camera. My point is that turning off the main camera and other cameras, actually, **can not** improve the performance. Thus, having it or not in headless mode is trivial. As I said before, the latency is caused by retrieving the image from GPU to RAM. As data in the chase camera buffer is always kept in GPU, so disable them won't improve too much efficiency.

Actually, I am wondering if we can modify the source code of our game engine, in particular, its rendering pipeline, and provide an API for directly manipulating the images on GPU, allowing torch/tf to directly access them. @pengzhenghao

MoreTore, 2023-02-16:
Good to know. That would be ideal.

MoreTore, 2023-02-16:
If i capture the main camera it is much faster. I am able to get over 40fps while sending the main window over ipc. I think it would be good to refactor the chase camera to lock to the widow of the car.

QuanyiLi, 2023-02-17:
Agree. I already merged a new PR into the main branch for allowing what you want. You could add the following parameters to env config to adjust the main camera position:
```
main_camera_config={
"camera_dist": -1,
"camera_pitch": 30,
"camera_height": 1,
"camera_smooth": False,
"show_interface": False,
}
```

MoreTore, 2023-02-17:
Wow thanks you

QuanyiLi, 2023-02-17:
@MoreTore I emerged several new patches, containing some small bug-fixing that may relate to your project. Please update your main branch.

Also, could you share your code for getting the main camera image with me? I want to know what causes that difference.

MoreTore, 2023-02-17:
Someone else is taking over. https://github.com/commaai/openpilot/pull/27373

QuanyiLi, 2023-02-17:
Cool man! I will go check it. Actually, we do need a decent driving system for our further research on scenario-based testing. We will try Openpilot.

MoreTore, 2023-02-17:
https://github.com/commaai/openpilot/issues/27284

jon-chuang, 2023-02-17:
@QuanyiLi , one small optimization is https://github.com/commaai/openpilot/blob/fdfebed860a3844b4fa87134c57e26e69f78ec08/tools/sim/bridge.py#L702

pengzhenghao, 2023-02-19:
@jon-chuang Your optimization is already merged in #300 

@MoreTore Your request to disable useless cameras in offscreen mode is merged in #305

closed 2023-02-19 by pengzhenghao