metadriverse/metadrive #292: np.float deprecated of numpy

CarlDegio, 2023-02-16:
I use pip install metadrive on windows, miniconda, python 3.9. But the metadrive.examples.profile_metadrive run error with report:
```
python -m metadrive.examples.profile_metadrive
Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
Start to profile the efficiency of MetaDrive with 1000 maps and ~8 vehicles!
Traceback (most recent call last):
  File "D:\program\miniconda3\envs\metadrive\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\program\miniconda3\envs\metadrive\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\examples\profile_metadrive.py", line 20, in <module>
    obs = env.reset()
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\envs\base_env.py", line 342, in reset
    self.engine.reset()
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\engine\base_engine.py", line 215, in reset
    manager.reset()
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\manager\map_manager.py", line 54, in reset
    map = self.spawn_object(PGMap, map_config=map_config, random_seed=None)
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\manager\map_manager.py", line 26, in spawn_object
    map = self.engine.spawn_object(object_class, auto_fill_random_seed=False, *args, **kwargs)
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\engine\base_engine.py", line 121, in spawn_object
    obj = object_class(**kwargs)
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\component\map\base_map.py", line 51, in __init__
    self._generate()
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\component\map\pg_map.py", line 51, in _generate
    self._big_generate(parent_node_path, physics_world)
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\component\map\pg_map.py", line 73, in _big_generate
    big_map.generate(self._config[self.GENERATE_TYPE], self._config[self.GENERATE_CONFIG])
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\component\algorithm\BIG.py", line 81, in generate
    if self.big_helper_func():
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\component\algorithm\BIG.py", line 89, in big_helper_func
    self._forward()
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\component\algorithm\BIG.py", line 132, in _forward
    block = self.sample_block()
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\component\algorithm\BIG.py", line 111, in sample_block
    block = block_type(
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\component\pgblock\pg_block.py", line 79, in __init__
    super(PGBlock, self).__init__(
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\component\block\base_block.py", line 31, in __init__
    super(BaseBlock, self).__init__(str(block_index) + self.ID, random_seed, escape_random_seed_assertion=True)
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\base_class\base_object.py", line 66, in __init__
    super(BaseObject, self).__init__(name, random_seed, config)
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\base_class\base_runnable.py", line 27, in __init__
    self.sample_parameters()
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\base_class\base_runnable.py", line 88, in sample_parameters
    self.update_config(ret)
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\base_class\configurable.py", line 28, in update_config
    self._config.update(config, allow_add_new_key=allow_add_new_key)
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\utils\config.py", line 139, in update
    self._update_single_item(k, v, allow_add_new_key)
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\utils\config.py", line 161, in _update_single_item
    self._set_item(k, v, allow_overwrite)
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\metadrive\utils\config.py", line 213, in _set_item
    if isinstance(value, (np.float32, np.float64, np.float)):
  File "D:\program\miniconda3\envs\metadrive\lib\site-packages\numpy\__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
```
consider to change version or change code?

QuanyiLi, 2023-02-16:
Hi,

How did u install metadrive? If you installed it from the cloned git repo, I think it should be fixed already. But if you installed via pip, it still happens as we don't release a new version yet. In this case, you can still solve it by ```pip install numpy==1.21.6```

Quanyi

pengzhenghao, 2023-02-16:
Yes, we have fixed this in latest main branch.

CarlDegio, 2023-02-17:
OK, I got that. I install metadrive by pip. I will try install by gie repo.

CarlDegio, 2023-02-17:
I reinstall it by git repo. the metadrive.examples.profile_metadrive run successfully. But another problem arised when I run 
`python -m metadrive.examples.drive_in_single_agent_env` and
 `python -m metadrive.examples.drive_in_safe_metadrive_env`:

```
python -m metadrive.examples.drive_in_single_agent_env
Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
Known pipe types:
  wglGraphicsPipe
(all display modules loaded.)
Traceback (most recent call last):
  File "D:\program\miniconda3\envs\metadrive\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\program\miniconda3\envs\metadrive\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "d:\pythonspace\metadrive\metadrive\examples\drive_in_single_agent_env.py", line 56, in <module>
    raise e
  File "d:\pythonspace\metadrive\metadrive\examples\drive_in_single_agent_env.py", line 36, in <module>
    o = env.reset()
  File "d:\pythonspace\metadrive\metadrive\envs\base_env.py", line 355, in reset
    self.lazy_init()  # it only works the first time when reset() is called to avoid the error when render
  File "d:\pythonspace\metadrive\metadrive\envs\base_env.py", line 250, in lazy_init
    engine = initialize_engine(self.config)
  File "d:\pythonspace\metadrive\metadrive\engine\engine_utils.py", line 11, in initialize_engine
    cls.singleton = cls(env_global_config)
  File "d:\pythonspace\metadrive\metadrive\engine\base_engine.py", line 28, in __init__
    EngineCore.__init__(self, global_config)
  File "d:\pythonspace\metadrive\metadrive\engine\core\engine_core.py", line 206, in __init__
    self.force_fps = ForceFPS(self)
  File "d:\pythonspace\metadrive\metadrive\engine\core\force_fps.py", line 20, in __init__
    self.engine.taskMgr.add(self.force_fps_task, "force_fps")
AttributeError: 'NoneType' object has no attribute 'taskMgr'
```

pengzhenghao, 2023-02-17:
Typically, this bug happens when you run two MetaDrive environments in the same process without closing the first environment. I would guess you are using Jupyter notebook to test the scripts? If that's the case you might need to restart the notebook before running the second script.

Please try this:

```
env1 = MetaDriveEnv(...)
env1.reset()
...
env1.close()

env2 = MetaDriveEnv(...)
env2.reset()
env2.step(...)
```

CarlDegio, 2023-02-17:
Actually, I test these scripts in cmd. So there are only one metadrive is running.  the report of #233 just crash with no message. But I have some run error.

pengzhenghao, 2023-02-17:
Thanks for reporting! Will look into it soon.

pengzhenghao, 2023-02-17:
I don't have linux/windows by my side so I have to run the script by disabling the native renderer. Everything look good (even with Pygame renderer).

CarlDegio, 2023-02-17:
I think the default render of metadrive on Windows is pygame? I run some test on pygame but no exception. I guess the self.engine have some problem so no variable? My metadrive version is 0.2.6.0.

QuanyiLi, 2023-02-17:
@CarlDegio @pengzhenghao 
I already fixed this in #293 . This bug is introduced in the last PR solving mutual reference. Please pull the latest main branch to catch up.  And the default renderer is Panda3D, while people could still use Pygame renderer on machines where OpenGL is not supported.

Quanyi

CarlDegio, 2023-02-17:
I got that, thank you!

CarlDegio, 2023-02-17:
By the way, I found the version of 3.10 python will lead to install fail(such as panda3d version). Maybe add a python version constraint will be better

QuanyiLi, 2023-02-17:
> By the way, I found the version of 3.10 python will lead to install fail(such as panda3d version). Maybe add a python version constraint will be better

Good suggestion. We are going to do a versioncheck in ```setup.py```

closed 2023-02-17 by QuanyiLi