metadriverse/metadrive #295: Auto drive mode switch crash on windows

CarlDegio, 2023-02-17:
I meet a bug of auto drive mode switch. I press T and then the program crash:
```
python -m metadrive.examples.drive_in_single_agent_env
Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
Known pipe types:
  wglGraphicsPipe
(all display modules loaded.)
Keyboard Shortcuts:
  W: Acceleration
  S: Braking
  A: Moving Left
  D: Moving Right
  R: Reset the Environment
  H: Help Message
  F: Switch FPS to unlimited / realtime
  Q: Third-person View Camera
  B: Top-down View Camera (control: WASD-=)
  +: Lift Camera
  -: Lower Camera
  Mouse click: move camera (top-down view)
  Esc: Quit

The observation is an numpy array with shape:  (261,)
<------here I press T-------->
The expert takeover is set to:  False
Traceback (most recent call last):
  File "D:\program\miniconda3\envs\metadrive\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\program\miniconda3\envs\metadrive\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\PythonSpace\metadrive\metadrive\examples\drive_in_single_agent_env.py", line 56, in <module>
    raise e
  File "D:\PythonSpace\metadrive\metadrive\examples\drive_in_single_agent_env.py", line 46, in <module>
    o, r, d, info = env.step([0, 0])
  File "D:\PythonSpace\metadrive\metadrive\envs\base_env.py", line 267, in step
    engine_info = self._step_simulator(actions)
  File "D:\PythonSpace\metadrive\metadrive\envs\base_env.py", line 294, in _step_simulator
    scene_manager_before_step_infos = self.engine.before_step(actions)
  File "D:\PythonSpace\metadrive\metadrive\engine\base_engine.py", line 290, in before_step
    new_step_infos = manager.before_step()
  File "D:\PythonSpace\metadrive\metadrive\manager\agent_manager.py", line 229, in before_step
    action = policy.act(agent_id)
  File "D:\PythonSpace\metadrive\metadrive\policy\manual_control_policy.py", line 57, in act
    return self.controller.process_input(self.engine.current_track_vehicle)
  File "D:\PythonSpace\metadrive\metadrive\engine\core\manual_controller.py", line 78, in process_input
    return np.array([self.steering, self.throttle_brake], dtype=np.float64)
ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (2,) + inhomogeneous part.
```
And press R(reset env) also crash.

Other key is ok, such as +/-/B/H/F. WASD can't test in auto mode so I can't test them.

QuanyiLi, 2023-02-17:
Hummm, I can not reproduce it. What is your ```numpy``` version? and Do you change anything in that script?

CarlDegio, 2023-02-17:
numpy version is 1.24.2, and I install metadrive just now so there are no change. Maybe I reinstall it

CarlDegio, 2023-02-17:
I reinstall it but still crash. It report some dimension problems.

CarlDegio, 2023-02-17:
I found this:
[image]
[image]

Actually, this two seems should be float without shape

CarlDegio, 2023-02-17:
seems problem here: in metadrive\engine\core\manual_controller.py
[image]
this rand lead to different shape.

QuanyiLi, 2023-02-17:
Yes, I got it. The numpy-1.24.2 causes this. Already fixed it and merged to main branch.

closed 2023-02-17 by QuanyiLi