metadriverse/metadrive #303: Error gym 0.24.1

jon-chuang, 2023-02-18:
```
/home/jonch/.cache/pypoetry/virtualenvs/openpilot-P9NmkXAX-py3.8/lib/python3.8/site-packages/gym/utils/passive_env_checker.py:97: UserWarning: WARN: We recommend you to use a symmetric and normalized Box action space (range=[-1, 1]) https://stable-baselines3.readthedocs.io/en/master/guide/rl_tips.html
  logger.warn(
/home/jonch/.cache/pypoetry/virtualenvs/openpilot-P9NmkXAX-py3.8/lib/python3.8/site-packages/gym/utils/passive_env_checker.py:216: UserWarning: WARN: Future gym versions will require that `Env.reset` can be passed a `seed` instead of using `Env.seed` for resetting the environment random number generator. 
  logger.warn(
/home/jonch/.cache/pypoetry/virtualenvs/openpilot-P9NmkXAX-py3.8/lib/python3.8/site-packages/gym/utils/passive_env_checker.py:228: UserWarning: WARN: Future gym versions will require that `Env.reset` can be passed `return_info` to return information from the environment resetting.
  logger.warn(
/home/jonch/.cache/pypoetry/virtualenvs/openpilot-P9NmkXAX-py3.8/lib/python3.8/site-packages/gym/utils/passive_env_checker.py:233: UserWarning: WARN: Future gym versions will require that `Env.reset` can be passed `options` to allow the environment initialisation to be passed additional information.
  logger.warn(
Known pipe types:
  glxGraphicsPipe
(all display modules loaded.)
:device(error): Error adding inotify watch on /dev/input: No space left on device
You may using too large buffer! The width is 1208, and length is 1928.
SELF 1928 1208
Process Process-1:
Traceback (most recent call last):
  File "/home/jonch/.pyenv/versions/3.8.10/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/jonch/.pyenv/versions/3.8.10/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "tools/sim/bridge.py", line 372, in bridge_keep_alive
    self._run(q)
  File "tools/sim/bridge.py", line 396, in _run
    world = self.spawn_objects()
  File "tools/sim/bridge.py", line 718, in spawn_objects
    env.step([0.0, 1.0])
  File "/home/jonch/.cache/pypoetry/virtualenvs/openpilot-P9NmkXAX-py3.8/lib/python3.8/site-packages/gym/wrappers/order_enforcing.py", line 37, in step
    return self.env.step(action)
  File "/home/jonch/.cache/pypoetry/virtualenvs/openpilot-P9NmkXAX-py3.8/lib/python3.8/site-packages/gym/wrappers/env_checker.py", line 39, in step
    return passive_env_step_check(self.env, action)
  File "/home/jonch/.cache/pypoetry/virtualenvs/openpilot-P9NmkXAX-py3.8/lib/python3.8/site-packages/gym/utils/passive_env_checker.py", line 273, in passive_env_step_check
    if np.any(np.isnan(obs)):
TypeError: ufunc 'isnan' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''
```

jon-chuang, 2023-02-18:
Maybe let's exclude gym 0.24.1 too

> Replaced the environment checker introduced in V24, such that the environment checker will not call step and reset during make. This new version is a wrapper that will observe the data that step and reset returns on their first call and check the data against the environment checker. @pseudo-rnd-thoughts

QuanyiLi, 2023-02-18:
Complete. Please check it again

jon-chuang, 2023-02-18:
Working for me with the following settings:
gym 0.24.0
panda3d 1.10.13

pseudo-rnd-thoughts, 2023-02-18:
We highly do not recommend using gym v0.24, rather you should use v0.22 or v0.25

QuanyiLi, 2023-02-18:
@pseudo-rnd-thoughts Done, use 0.22 instead

PR #302 (2023-02-21): Update dependency version
## What changes do you make in this PR?

    "gym>=0.20.0, <=0.24.0",
    "numpy>=1.21.6, <=1.24.2",
    "python>=3.6, <12",
    "panda3d>=1.10.8, <=1.10.13"

close #299  
close #301 
close #286
close #303 

Unitest Pass with Panda=1.10.13, numpy=1.24.2, gym=0.24.0, python-3.11:
- [x] Linux
- [x] Windows
- [x] Mac  

@pengzhenghao Could you help test python-3.11 on Mac? Just run all unit test locally

## Checklist

* [x] I have merged the latest main branch into current branch.
* [x] I have run `bash scripts/format.sh` before merging.
* Please use "squash and merge" m [... cut]
closed 2023-02-21 by QuanyiLi