metadriverse/metadrive #306: Access rendered image on GPU via Torch

QuanyiLi, 2023-02-20:
I learned that it is feasible to accelerate camera efficiency by accessing the rendered image on GPU without moving to RAM:

**Reference:**
1. https://discourse.panda3d.org/t/allowing-torch-tensorflow-to-directly-access-rendered-image-in-gpu/29119/2
2. https://discourse.panda3d.org/t/how-to-get-gpu-memory-pointer-from-texture-object/27983/3
3. #290 

I would like to try this, as visuomotor policy training is one drawback for MetaDrive.

QuanyiLi, 2023-02-25:
I make it!

Running: ```/home/shady/data/metadrive/metadrive/tests/vis_functionality/vis_cuda_image.py```

FPS Before optimization: 1100~
FPS After optimization: **4000+**

QuanyiLi, 2023-02-25:
@jon-chuang 
Would you like to accelerate image retrieval? I just tested a new feature that supports accessing the rendered image without copying it to the host memory. All things happen on GPU now!

QuanyiLi, 2023-02-26:
## New test results in MetaDrive Env

Script: ```metadrive/tests/vis_functionality/vis_profile_metadrive_cuda.py```

Setting: 
- 3 block
-  no traffic
-  agent policy: IDM policy
- Image size: 1920 * 1080

Result:
Ram Image (old): 25 FPS
CUDA Image (new): **385 FPS** 

Amazing!

pengzhenghao, 2023-02-27:
Can only run `vis_cuda_make_buffer.py` with this:

[image]

All other `vis_cuda` fail with:

[image]

Might caused by wrongly installed cupy in Windows. I can't install it through `pip install cupy-cuda12x` and can only install via `conda install -c conda-forge cupy`

QuanyiLi, 2023-02-27:
I have already 
1. Fixed this problem on windows
2. Added installation instructions to Doc. 

Please check the Installation section, follow the guide and help me verify the installation pipeline and check my writing.

jon-chuang, 2023-02-27:
Amazing! Thanks for working on this.

Unfortunately, we require the image buffer to be in memory. Furthermore, the target we'd like to optimize most for is CPU-only setups.

closed 2023-02-27 by QuanyiLi

mimi89999, 2024-04-09:
Hello,
Would it be possible to do the same thing with OpenCL?

QuanyiLi, 2024-04-15:
@mimi89999 I think so. Just search the keyword: OpenGL OpenCL interop python. There are many discussions like this: https://cyrille.rossant.net/a-tutorial-on-openglopencl-interoperability-in-python/#:~:text=With%20OpenCL%2DOpenGL%20interoperability%2C%20one,GPU%20memory%20at%20all%20times.

But for OpenCL to Torch, I am not sure if there is some tools helping with it. (Though I know torch has OpenCL backend)

PR #312 (2023-02-27): Convert OpenGL rendered image to Torch tensor
## What changes do you make in this PR?

close #306 
close #313

## Checklist

* [ ] I have merged the latest main branch into current branch.
* [ ] I have run `bash scripts/format.sh` before merging.
* Please use "squash and merge" mode.