metadriverse/metadrive #308: Can I set up discrete actions?

shile1998, 2023-02-22:
Can I set up discrete actions?

QuanyiLi, 2023-02-22:
Yes. The simple way is to set ```discrete_action=True``` when creating your env with a config. But I am not sure if it works or not. Or you can write a customized env wrapper and parse the discrete action index to continuous action, like 1 for [1, 0], 2 for [-1,0], 3 for [1,1], 4 for [1. -1] and so on.

pengzhenghao, 2023-02-22:
Yes you can use `config["discrete_action"] = True` to enable discrete action space.

But for now we only support MultiDiscrete action space. 

I am working on adding a knob to control whether to use MultiDiscrete or Discrete, maybe release in today.

pengzhenghao, 2023-02-22:
I am introducing a new config called "use_multi_discrete", so you can control whether you wish to use Discete or MultiDiscrete action spaces.

Now the config relating to discrete action are:

```python
config = dict(
    discrete_action=False,
    discrete_steering_dim=5,
    discrete_throttle_dim=5,
    # When discrete_action=True: If True, use MultiDiscrete action space. Otherwise, use Discrete.
    use_multi_discrete=False,
)
```

When using Discrete action space, your action space will takes an integer between `[0, discrete_steering_dim * discrete_throttle_dim)` .

When using MultiDiscrete action space, your action will takes two integers `[0, discrete_steering_dim ) and [0, discrete_throttle_dim)`.

MetaDrive will convert whaterver input to two [-1, 1] continuous values just like in continuous action space.

shile1998, 2023-02-23:
Thanks for replying, can the discrete action be simpler, such as discrete steering set to 2, corresponding to left and right, respectively, and then the vehicle will drive in the middle of the left or right side, and directly change lanes

pengzhenghao, 2023-02-23:
You means you want a low-level controller that can automatically conduct lane-chaniging? So the action space remains 3 valid actions: changing lane to left, to right and lane keeping?

@QuanyiLi I think our IDM policy has the function for these, but is it really capable to serve as a low-level controller? In complex scenario such as intersection, I doubt whether turning left / right roads can be achieved by these 3 actions.

shile1998, 2023-02-23:
I hope to get some advice, I want to complete the function of single-agent overtaking lane change, how to achieve a fluent lane change like a promotional video, such as algorithm selection, whether to choose discrete action algorithm or continuous, about how long to train, and what to pay attention to

QuanyiLi, 2023-02-23:
I can help you with it. We have some built-in driving policies, which can be modified to provide these primitive lane-changing actions. It won't take a long time.

shile1998, 2023-02-23:
Can that be elaborated

shile1998, 2023-02-23:
Probably you said PPO

QuanyiLi, 2023-02-23:
Nah, we actually have an IDM model that can perform lane changing when it detects static or slow-moving obstacles in front of the vehicle. Based on it, I can design a new action space, let's say, two discrete actions, where action 0 for changing to left, 1 for right. After receiving the action from ```env.step()```, the lane changing will be performed by the low-level controller automatically.

shile1998, 2023-02-23:
Ok, thank you for your reply. I will try to change it. During this period, it would be better if you could give the tutorial

QuanyiLi, 2023-02-23:
I can help you build it. In this case, you can focus on your own project, maybe the RL training stuff without touching the low-level Metadrive things. Also, we will merge this into the main branch as a feature. I will finish this soon.

QuanyiLi, 2023-02-24:
Hi,@shile1998 

I provide this feature now. You can find a demo by running this file as script: https://github.com/metadriverse/metadrive/blob/discrete-lane-change-policy/metadrive/tests/test_policy/test_lane_change_policy.py

shile1998, 2023-02-24:
After running, there are some problems, there will be many debug errors in the terminal, and the simulation interface is blank

QuanyiLi, 2023-02-24:
That is the debug mode, actually. Please pull the latest update and run it again!

QuanyiLi, 2023-02-25:
Hi, I merged this PR into ```main``` now. 

Feel free to ask about anything regarding DQN training or request new MetaDrive features!

PR #310 (2023-02-25): Discrete lane change policy
## What changes do you make in this PR?

Add features for #308 
close #308

## Checklist

* [ ] I have merged the latest main branch into current branch.
* [ ] I have run `bash scripts/format.sh` before merging.
* Please use "squash and merge" mode.
closed 2023-02-25 by QuanyiLi