metadriverse/metadrive #315: Visualize the problem after ray training

shile1998, 2023-02-28:
After EGPO training, I want to display the environment in metadrive, can you provide the corresponding code, I am not familiar with ray

shile1998, 2023-02-28:
Visualization of the most basic SAC algorithm

pengzhenghao, 2023-02-28:
Thanks for reaching

A basic logic is:

1. Instantiate the "Trainer" instance from RLLib with proper config.
2. Restore checkpoint
3. Create environment
4. Rollout the environment and probably render / record the visualization.

```python

# Create and restore your agent
from ray.rllib.agents.sac import SACTrainer
...

trainer = SACTrainer({some config identical to your training config})
trainer.restore(CHECKPOINT_FOLDER_PATH)

...

# Create environment
env = SafeMetaDrive({some env config})
...

# Rollout
action = trainer.compute_single_action(obs)
o, r, d, i = env.step(action)

...
```

closed 2023-04-23 by QuanyiLi