metadriverse/metadrive #319: Can metadrive be used in pymarl? Is there any code examples?

iamrice, 2023-03-07:
I notices that the related work in your group are developed based on ray framework. Can you please compare the advantages or disadvantages between pymarl and ray when using metadrive?

I am new to RL and struggled when selecting frameworks. Very thanks.

pengzhenghao, 2023-03-08:
Don't recommand RLlib if you are new to RL. 

Pymarl is good but it seems that it does not support varying the number of agents. In MARL envs in MetaDrive the number of vehicles are keep changing all the time. (Though it is possible to maintain the number of agents to a fixed value during one episode, but this is not the default behavior of MetaDrive MARL envs) Therefore I don't think you can connect Pymarl and MetaDrive immedetely without modification. I would be happy to implement that functionality if you do really need it.

For the multi-agent framework, I came across a repo MA-RLLib:  https://github.com/Replicable-MARL/MARLlib  which seems to support MetaDrive MARL envs. But they don't implement CoPO algorithm.

We are turning to stable-baselines 3 when implementing single-agent RL algorithm, due to its flexibity. But SB3 does not have well support for MARL envs.

I believe there are much more great frameworks that I am unfortunately to have try on. Please tell me if you find the best of you!

iamrice, 2023-03-08:
Thanks. But I wonder why don't you recommand ray-RLlib? 🤣 Actually I have read the source code of "copo" for several days. And I think it may be a good way to validate one of my ideas. 🤣Maybe I will try it.

Regrading pymarl, I found that pymarl has stopped updating RL algorithms two years old. So I don't think it is convinient to compare my algorithm to sota algorithms. What do you think? 🤔

Also appreciate your recommendation of MA-RLLib. I think it would be helpful to my future  research because it contains many environments.

pengzhenghao, 2023-03-10:
You can take a try on modifying CoPO code. The torch version of it is much more readable. Remember to setup `local_mode=True` so you can pause at breakpoints to see data flow.

closed 2023-04-23 by QuanyiLi