metadriverse/metadrive #332: 'done' is always true starting from first step in MARL env

jiw18020, 2023-03-13:
Hi, 

I'm trying to use the MARL env, and here is my code, which is copied from the document, I printed the done and it's true every step, that's really weird to me, can anyone help with this issue? Thanks a lot for your time! 

CLASS_NAME = 'tollgate'
from metadrive import (
    MultiAgentMetaDrive,
    MultiAgentTollgateEnv,
    MultiAgentBottleneckEnv,
    MultiAgentIntersectionEnv,
    MultiAgentRoundaboutEnv,
    MultiAgentParkingLotEnv
)

envs_classes = dict(
    roundabout=MultiAgentRoundaboutEnv,
    intersection=MultiAgentIntersectionEnv,
    tollgate=MultiAgentTollgateEnv,
    bottleneck=MultiAgentBottleneckEnv,
    parkinglot=MultiAgentParkingLotEnv,
    pgma=MultiAgentMetaDrive
)
env = envs_classes[CLASS_NAME]()

for i in range(1000):
    obs, reward, done, info = env.step(env.action_space.sample())  # Use random policy
    #print(env.action_space.sample())
    if i%100 == 0:
        print(reward)
    #env.render()
    if done:
        print(i)
        print('done')
        env.reset()
env.close()

QuanyiLi, 2023-03-13:
well, I think ```done``` here is a ```dict``` containing the termination status of all agents. You should query the status for each agent via their id like ```agent0```. For getting a global termination status, check ```done["__all__"]``` instead

jiw18020, 2023-03-13:
Hi Quanyi,

Thanks for your reply! I just found out that done is a dict for all agents. I have a following question, as the default length is 1000 in MARL env, after 1000 steps, the vehicles that arrive earlier at their destination will terminate and the vehicles that haven't arrived will keep running, is that correct? Also, I noticed that lots of projects, like Copo, use RLlib to train the MARL algorithm, are there any repos directly train the MARL on Metadrive without using RLlib?

QuanyiLi, 2023-03-14:
Hi, Zhenghao@pengzhenghao, could you help with it?

pengzhenghao, 2023-03-14:
> as the default length is 1000 in MARL env, after 1000 steps, the vehicles that arrive earlier at their destination will terminate and the vehicles that haven't arrived will keep running, is that correct?

No. After 1000 steps, the environment stop respawning new vehicles. But the existing vehicles are still alive and will be terminated if their own horizons are reached.

There are 3 config to control this behavior:

config["max_step_per_agent”] Default to 1000. means the maximum number of the steps of each agent.

config[“horizon”] Default to 1000. is the maximum length of the episode. If allow respawn, then this is the maximum step that respawn can happen. After that, the episode won't terminate until all existing vehicles reach their horizon or done.

config[“allow_respawn”]  Default to True.

> I noticed that lots of projects, like Copo, use RLlib to train the MARL algorithm, are there any repos directly train the MARL on Metadrive without using RLlib?

MA-RLLib support MetaDrive.

Best,
Zhenghao

jiw18020, 2023-03-14:
Hi Zhenghao,

Thanks for the reply!

closed 2023-03-14 by jiw18020