metadriverse/metadrive #335: Hard to integrate the RAY/RLLib into metadrive

siaoliu, 2023-03-15:
There is a dependency about 'protobuf==3.20.1' in metadrive;
But the ray/RLLib(1.2.0) requires 'protobuf >= 3.20.3', which is hard to address.

I try to address it with another codebase 'https://github.com/decisionforce/metadrive'.

QuanyiLi, 2023-03-15:
This post says: https://stackoverflow.com/questions/12759761/pip-force-install-ignoring-dependencies
> pip has a --no-dependencies switch. You should use that.
For more information, run pip install -h, where you'll see this line:
--no-deps, --no-dependencies

This allows you to ignore the conflicts by ```pip install protobuf == 3.20.1 --no-deps```

pengzhenghao, 2023-03-15:
Do we really need to specify the version of protobuf ？What is this package used for? @QuanyiLi

QuanyiLi, 2023-03-16:
Waymo-related things require it. Maybe some data processing tools. Consider removing it later.

closed 2023-04-23 by QuanyiLi