metadriverse/metadrive #338: maybe some errors of state infomation

iamrice, 2023-03-16:
I create an environment and run:
```
MultiAgentIntersectionEnv(dict(num_agents=30, crash_done=True, use_render=False))
```

at the first step, i print the keys of observation:
```
dict_keys(['agent0', 'agent1', 'agent2', 'agent3', 'agent4', 'agent5', 'agent6', 'agent7', 'agent8', 'agent9', 'agent10', 'agent11', 'agent12', 'agent13', 'agent14', 'agent15', 'agent16', 'agent17', 'agent18', 'agent19', 'agent20', 'agent21', 'agent22', 'agent23', 'agent24', 'agent25', 'agent26', 'agent27', 'agent28', 'agent29'])
```
after several steps, i got:
```
dict_keys(['agent0', 'agent1', 'agent2', 'agent3', 'agent5', 'agent6', 'agent7', 'agent8', 'agent9', 'agent10', 'agent11', 'agent12', 'agent13', 'agent14', 'agent15', 'agent17', 'agent18', 'agent19', 'agent20', 'agent21', 'agent22', 'agent23', 'agent24', 'agent25', 'agent26', 'agent27', 'agent28', 'agent29', 'agent30'])
```

there is an extra key "agent30". I don't know should I use agent0-agent29 or agent1-agent30?

iamrice, 2023-03-16:
I guess new vehicles will be added into the environment when original vehicles crashed, right?

Can I cancle this setting? Just let the number of vehicles decrease when crash occurs.

iamrice, 2023-03-16:
Or, can I set a maximum number of the vehicles? For example, put 30 vehicles in the environment at the begining, and gradually add new vehicles util 100 vehicles (including vehicles that have crashed).

iamrice, 2023-03-16:
I reconstruct the "_respawn_vehicles" function and solve it.

pengzhenghao, 2023-03-16:
> I guess new vehicles will be added into the environment when original vehicles crashed, right?

Correct.

> Can I cancle this setting? Just let the number of vehicles decrease when crash occurs.

Yes, set

`config["allow_respawn"]=False`

> can I set a maximum number of the vehicles? 

Yes. 

`config["num_agents"] = 30` does this.

> For example, put 30 vehicles in the environment at the beginning, and gradually add new vehicles util 100 vehicles (including vehicles that have crashed).

Sorry. It seems that we don't support creating less agents in the beginning and gradually increasing it since when to introduce "extra" agents is underdetermined.

---

In case you don't know what happen, I will try to describe the logic behind.

MetaDrive creates  `config["num_agents"]` in the beginning of each episode. If `config["allow_respawn"]=True`, then when one vehicle is done, a new vehicle immediately respawn at random spawn position. So the total number of vehicles are kept to `config["num_agents"]`.

When `config["horizon"]` is hit, the respawn stops. But existing vehicles are still running in the environment  until all of them are done.

---

Could you please tell what modification you make to `_respawn_vehicles`? I don't think you need to modify the source code. If does, then maybe we should introduce your change to MetaDrive.

pengzhenghao, 2023-03-16:
For a special case when user want to access the "just terminated" vehicle, you can use

```
env.vehicles_including_just_terminated()
```

instead of 

```
env.vehicles()
```

pengzhenghao, 2023-03-16:
The return keys in observation are always the keys of existing vehicles (that is `env.vehicles()`)

iamrice, 2023-03-16:
Thank you very much. I realize my requirement in this way😂:
```
def _respawn_vehicles(self, randomize_position=False):
            new_obs_dict = {}
            new_info_dict = {}
            if not self.agent_manager.allow_respawn:
                return new_obs_dict, new_info_dict
            **while(self.agent_manager.next_agent_count < self.maximum_vehicles):**
                new_id, new_obs, step_info = self._respawn_single_vehicle(randomize_position=randomize_position)
                if new_obs is not None:
                    new_obs_dict[new_id] = new_obs
                    new_info_dict[new_id] = step_info
                else:
                    break
            return new_obs_dict, new_info_dict
```
maximum_vehicles is an extra key I add.

pengzhenghao, 2023-03-26:
Oh I know what you are doing. I guess you are using some RL library that requires a fixed number of agents in MARL env? So your code here enforce the environment to spawn agents until the required number of environments exists in the env.

There is a potential bug, that is there are not enough empty spawn slots exist in the map so the system can't find enough places to spawn new vehicles. 

If this create bugs for you, I would like to add some fixs to it or even integrate your code into MD.

iamrice, 2023-04-03:
Thanks. What you guessed is one of the reasons. Another reason is that I want to evaluate a policy with less steps.

However, I don't think this is a potential bug. If there are not enough empty spawn slots exist in the map, the function "_respawn_single_vehicle" will return "none".

iamrice, 2023-04-03:
@pengzhenghao I have some doubts about the experiment results of metadrive and Copo, which is issued in https://github.com/decisionforce/CoPO/issues/36#issuecomment-1482175336.

Have you noticed it? I would really appreciate it if you could clear up my confusion.

closed 2023-04-23 by QuanyiLi