metadriverse/metadrive #346: Kickoff the integration of RenderPipeline

pengzhenghao, 2023-03-17:
We should first understand the installation workflow and create submodule requirement in setup.py for this package.

https://github.com/tobspr/RenderPipeline

QuanyiLi, 2023-03-17:
I will try making a minimal runnable example before deciding whether to integrate it or not

QuanyiLi, 2023-03-18:
The rendering quality is good, while the are still other problems, like some car models can not be rendered correctly. Road and terrain has no *Normal texture* so always dark. The motion blur seems jitter. 

We should do this integration wisely, keeping the most important functions we want and discarding those causing overhead.

[image]

QuanyiLi, 2023-03-19:
@pengzhenghao Checkout this video
These clouds are really beautiful. As you can see, some models and textures (except ego car) are rendered incorrectly. 

Thus, this is not that easy to do a quick integration. We need a standard asset-building pipeline that can make sure models edited in blender can be shown correctly. Also, this repo includes some fancy debug tools for adjusting the rendering effect on-the-fly. 
https://user-images.githubusercontent.com/24489294/226180810-782cb0fc-14d9-446e-896b-fcc9d1657c4e.mp4

QuanyiLi, 2023-03-19:
BTW, some CG stuff is needed for understanding the render pipeline's design philosophy... We have to take a look at some basic CG things.

closed 2023-04-23 by QuanyiLi