metadriverse/metadrive #379: Discussion: Would road sampling interval affects the road network, observation, navigation sys, reward?

pengzhenghao, 2023-03-27:
In Waymo Open Dataset information: https://waymo.com/intl/en_us/open/download/  It says the v1.2 version: 

> Increased the max number of map points in tf_examples to 30k and reduced sampling to 1.0m to increase map coverage, so the coverage equalizes that of the dataset in scenario proto format.

I wonder if the road sampling rate is changed, would that make the length of road segment in MetaDrive vary? 

If so, then the navigation checkpoints will become more denser? Reward function changes its scale? 

I don't know how much this affect the system.

**Please note that this discission is not only limited in the change between Waymo v1.1 and v1.2. What if NuPlan dataset has different road sampling rate? How to normalize this problem?**

QuanyiLi, 2023-03-27:
As the sampling rate affects the restored map structure, it will definitely affect the obs/reward and other things. But I think it would not cause too much difference if the sampling interval is lower than 1m.

pengzhenghao, 2023-03-27:
TrafficGen use 0.5m. I am not sure the sampling interval in Nuplan. Maybe we can introduce some interpolation (we already have??) to make sure we always stick to 1m sampling interval?

QuanyiLi, 2023-03-27:
When do exportation? We sample all maps in 0.2m intervals, regardless of the sampling rate of the original map. All original lane will be processed by Interpolating line

closed 2023-04-23 by QuanyiLi