metadriverse/metadrive #380: Discussion: Object Type formalization

pengzhenghao, 2023-03-27:
This is little discussion on the design choice. I saw we are using this to register different agent type. We also have another kind of objects in the environment, like Traffic Cone, Traffic Light and so on.

Here are my questions:

1. Should we stick to STRING to describe all objects instead of using Int?
2. Should we introduce a large "ObjectTypeClass" to make all objects within one class?

[image]

pengzhenghao, 2023-03-27:
Also, this class falls into "Waymo Type". We should upgrade it to a global things, like put it in MetaDrive.constants

pengzhenghao, 2023-03-27:
metadrive.constants defines some. But this is clearly mismatched from Waymo type names

[image]

QuanyiLi, 2023-03-27:
1. yes
2. what do you mean by **all**, you mean objects related to scenarios exportation? or all metadrive object. If it is the former one, you can **not** merge them. As in waymo original format, number is used to represent type, 1 might be vehicle for AgentType but yellow_line for LineType.

pengzhenghao, 2023-03-27:
@QuanyiLi Here is what I am thinking:

First, I am talking about all places. We should use a predefined set of strings to discribe all object types. And we will have a class to register this:

Option 1: A big class to store all types:

```python
# In metadrive.constants
class ObjectType:
    VEHICLE = "VEHICLE"
    TRAFFICLIGHT = "TRAFFICLIGHT"
    YELLOW_SOLID_LANE = "YELLOW_SOLID_LANE"
```

Option 2: Still using current method, but never use int in our code. It is OK to have a mapping function to transform from real dataset:
```python
# In metadrive.constants
class AgentType:
    VEHICLE = "VEHICLE"
    CYCLIST = "CYCLIST"

    @classmethod
    def from_waymo(waymo_agent_type: int):
        return {1: self.VEHICLE, 2: self.CYCLIST}
```

Second, when exporting data in the data format, still using the same string to describe the objects and never use any interger.

closed 2023-04-23 by QuanyiLi