metadriverse/metadrive #387: Sometimes atan2 sometimes -atan2

pengzhenghao, 2023-03-29:
What is the principle between this?

from abs_lane.py:

[image]

QuanyiLi, 2023-03-29:
have no idea. At least it has nothing to do with the bug you encountered. I guess it is related to converting to panda coordinates

QuanyiLi, 2023-03-29:
The following function uses it for setting a panda heading without times a ```-```. I am 99% sure it is an implicit heading convert. I already fixed it by calling explicit converting API.

QuanyiLi, 2023-03-29:
I ran all tests, and find new failures caused by 
1. ```ReplayEgoCarPolicy```:  
action = self.traj_info[int(self.timestep)].get("action", None)
E       IndexError: list index out of range

2. ```def replay_frame()```:
for name, config in self.current_frame.spawn_info.items():
E       AttributeError: 'list' object has no attribute 'spawn_info'

pengzhenghao, 2023-03-30:
This issue is resolved by introducing a general "panda_heading" function to transform heading angle.

closed 2023-03-30 by pengzhenghao