metadriverse/metadrive #421: A problem in run with ray

beniciolee, 2023-04-16:
When i run "python -m metadrive.examples.train_generalization_experiment",the error infomation is like fellow:
(metadrive) PS D:\metadrive> python -m metadrive.examples.train_generalization_experiment
Traceback (most recent call last):
  File "D:\anaconda3\envs\metadrive\lib\runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "D:\anaconda3\envs\metadrive\lib\runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "D:\metadrive\metadrive\__init__.py", line 1, in <module>
    import metadrive.register
  File "D:\metadrive\metadrive\register.py", line 1, in <module>
    from gym.envs.registration import register, registry
  File "D:\anaconda3\envs\metadrive\lib\site-packages\gym\__init__.py", line 15, in <module>
    from gym.envs import make, spec, register
  File "D:\anaconda3\envs\metadrive\lib\site-packages\gym\envs\__init__.py", line 5, in <module>
    _load_env_plugins()
  File "D:\anaconda3\envs\metadrive\lib\site-packages\gym\envs\registration.py", line 321, in load_env_plugins
    fn = plugin.load()
  File "D:\anaconda3\envs\metadrive\lib\site-packages\importlib_metadata\__init__.py", line 210, in load
    module = import_module(match.group('module'))
  File "D:\anaconda3\envs\metadrive\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "D:\anaconda3\envs\metadrive\lib\site-packages\ale_py\gym.py", line 5, in <module>
    from ale_py.roms.utils import rom_name_to_id, rom_id_to_name
  File "D:\anaconda3\envs\metadrive\lib\site-packages\ale_py\roms\__init__.py", line 89, in <module>
    ROMS = resolve_roms()
  File "D:\anaconda3\envs\metadrive\lib\site-packages\ale_py\roms\__init__.py", line 40, in resolve_roms
    supported, unsupported = package.resolve()
  File "D:\anaconda3\envs\metadrive\lib\site-packages\ale_py\roms\utils.py", line 60, in resolve
    lambda file: file.suffix == ".bin", resources.files(self.package).iterdir()
  File "D:\anaconda3\envs\metadrive\lib\site-packages\importlib_resources\_common.py", line 46, in wrapper
    return func(anchor)
  File "D:\anaconda3\envs\metadrive\lib\site-packages\importlib_resources\_common.py", line 56, in files
    return from_package(resolve(anchor))
  File "D:\anaconda3\envs\metadrive\lib\functools.py", line 820, in wrapper
    return dispatch(args[0].__class__)(*args, **kw)
  File "D:\anaconda3\envs\metadrive\lib\site-packages\importlib_resources\_common.py", line 82, in _
    return importlib.import_module(cand)
  File "D:\anaconda3\envs\metadrive\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "D:\anaconda3\envs\metadrive\lib\site-packages\atari_py\__init__.py", line 1, in <module>
    from .ale_python_interface import *
  File "D:\anaconda3\envs\metadrive\lib\site-packages\atari_py\ale_python_interface.py", line 18, in <module>
    'ale_interface/ale_c.dll'))
  File "D:\anaconda3\envs\metadrive\lib\ctypes\__init__.py", line 434, in LoadLibrary
    return self._dlltype(name)
  File "D:\anaconda3\envs\metadrive\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] 

where is the problem in installing ray or metadrive

beniciolee, 2023-04-16:
before i install ray==1.2.0 , run metadrive.examples right ,but after installing ray, the error occures

QuanyiLi, 2023-04-16:
It must be something wrong with ```gym```. Try ```pip install gym==0.19.0```

closed 2023-04-23 by QuanyiLi