metadriverse/metadrive #425: Update webpage github page

QuanyiLi, 2023-04-21:


QuanyiLi, 2023-10-25:
@pengzhenghao @BoSmallEar @WeizhenWang-1210 @pimpale 

I plan to finish the new documentation before the end of this year. It will cover the content I mentioned above. Do you have any suggestions?

CarlDegio, 2023-11-11:
1. the section "Read Data from Dataset" should be updated. This part is implemented through Scenarionet.
2. It is recommended to give the calculation process of vehicle dynamics simulation. Convenient for users to develop speed or trajectory control. Also for 'random_agent_model' in config.
3. DI-drive now give an exmaple about metadrive reinforcement learning. Can be added to the ecosystem.

CarlDegio, 2023-11-21:
4. Introduction of expert policy (expert_takeover) can be added. And how to train a policy like this.