metadriverse/metadrive #428: TypeError: 'NoneType' object is not subscriptable

AHPUymhd, 2023-04-23:
[image]
[image]
why!!!

QuanyiLi, 2023-04-23:
Could you give me more information about your setting? Like how do you initialize your environment? I did some simple tests on the latest MetaDrive, but can not reproduce it.

AHPUymhd, 2023-04-24:
initialize  environment is metadrive.envs.metadrive_env.py and I use MetaDrive-0.3.0.1 version  ,it maybe a problem of my conda?

QuanyiLi, 2023-04-25:
Sorry, I can't reproduce your problem. It seems that the bug is not on our side. I tested with the following lines of code.
[image]

QuanyiLi, 2023-04-25:
Hmmm, I have no idea about your situation. It works great on my machine. Maybe it is due to your gym version? I use ```gym==0.25.2```

[image]

closed 2023-04-26 by QuanyiLi