metadriverse/metadrive #432: how to control vehicle  (render = Ture)

AHPUymhd, 2023-04-24:
May I ask how to control vehicle?    env.vehicle.set_velocity([1, 0.1], 10) can set vehicle's velocity directly  ,but how to set steer and acceleration ? I have searched metadrive.tests   but didn't find this function   please teach me ,thinking you very much !!!!!!!!

QuanyiLi, 2023-04-25:
Just use ```env.step([Steering, throttle])``` to control the vehicle with acceleration and advance the simulation.

AHPUymhd, 2023-04-25:
[image]
[image]

AHPUymhd, 2023-04-25:
but the vehicle  didn't move ,what's my problem?

QuanyiLi, 2023-04-25:
in config set manual_control=False

发自我的 iPhone

closed 2023-04-26 by QuanyiLi