metadriverse/metadrive #444: [Visualize Bug] AssetLoader can not be properly initialized and fail to load "box.bam"

Zi-ang-Cao, 2023-05-25:
Hey Metadrive team, 
> The version of MetaDrive: 0.3.0.1. 

<b>I am trying to visualize the checkpoints (Nodes) in your marl_envs.</b>
While simply turning on both `self._show_navi_info` and `show_navi_mark` flags in `BaseNavigation` Class,  the following line never works.
```Python
# in the `metadrive/component/vehicle_navigation_module/base_navigation.py`
if show_navi_mark:
    navi_point_model = AssetLoader.loader.loadModel(AssetLoader.file_path("models", "box.bam"))
    navi_point_model.reparentTo(self._goal_node_path)

# Error Message:
# AttributeError: 'NoneType' object has no attribute 'loadModel'
```
I tried to add `AssetLoader.init_loader(self.engine)` before calling `env.engine.reset()`, but the error message is still there. 

Meanwhile, I found that so many visualizing functions might not be maintained in the current MetaDrive? (or at least, the declaration of those functions can not being found...)
```Python 
# Can not found the implementation of .setPos() and .setH() functions
self._goal_node_path.setPos(panda_vector(pos_of_goal[0], pos_of_goal[1], 1.8))
self._goal_node_path.setH(self._goal_node_path.getH() + 3)
```

Hope to hear from you guys soon!

QuanyiLi, 2023-05-27:
Hi, Zi-ang,

Sorry for the late reply. Are you using the 2D top-down renderer? The Navi mark is only available using the 3D renderer.

Quanyi

Zi-ang-Cao, 2023-05-28:
>

Zi-ang-Cao, 2023-05-28:
Hey Quanyi, 
Yes, you are right. I am using the 2D top-down renderer. Thanks for your reply!

Zi-ang-Cao, 2023-05-28:
Hey Quanyi, 
Yes, you are right. I am using the 2D top-down renderer. Thanks for your reply! 
Btw, do you have any suggestion for adding the Navi mark in the 2D top-down render?

Zi-ang-Cao, 2023-05-28:
> Hi, Zi-ang,
> 
> Sorry for the late reply. Are you using the 2D top-down renderer? The Navi mark is only available using the 3D renderer.
> 
> Quanyi

Hey Quanyi, 
Yes, you are right. I am using the 2D top-down renderer. Thanks for your reply! 
Btw, do you have any suggestion for adding the Navi mark in the 2D top-down render?

QuanyiLi, 2023-05-28:
Yes. I believe you can do some customization in ```class TopDownRenderer```. Just refer to the code for drawing the vehicle, which tells you how to draw a box in the scene. Then every frame you can access the checkpoint position from vehicle.navigation and draw a box representing the checkpoint.

To be specific, you should modify this function:
https://github.com/metadriverse/metadrive/blob/4d77e619b2bc53397c6836e26a17dfc43b5d8cd8/metadrive/obs/top_down_renderer.py#L339

Just write a new ```for``` loop or modify an existing one to traverse all vehicles so that you can get the navigation point position for each vehicle and use a similar way to draw it on the 2D renderer.

closed 2023-06-18 by Zi-ang-Cao