metadriverse/metadrive #448: How to implement trajectory planning on metadrive?

CarlDegio, 2023-06-04:
Under normal circumstances, trajectory planning gives the desired trajectory for the next 8 seconds or the current desired speed. Is there currently a tool that translates desired trajectory or speed to throttle and steering? 

I found that apollo or autoware have these functions, but they are too heavy. And control method such as PID depends on different car models. Do you have any good suggestions for implementing trajectory planning in metadrive or Carla?

QuanyiLi, 2023-06-07:
If you only want to study the trajectory planning problem with perfect control assumption, you cab directly set the vehicle positions, velocities, and headings to the target point. But if you want to consider the control error, you do need to convert trajectory to throttle and steering. However, unfortunately, we don't provide that, you have to implement it on your own.

Besides, for trajectory planning, I would recommend to check Casadi: https://github.com/casadi/casadi/blob/main/docs/examples/python/race_car.py

CarlDegio, 2023-06-07:
Thank you for your answer! i will try it

CarlDegio, 2023-06-08:
I want to represent the planned trajectory with lines on the ground. Is it possible to add a debug line on the ground? (Perhaps it needs to be plotted in panda3d?)

QuanyiLi, 2023-06-09:
Ah, for adding a line segment, you can use the ```line_np = env.engine.add_line(start, end, color, thickness)```. It will return the line object, then you can use ```line_np.reparentTo(env.engine.render)``` to set it at the world coordinates. Everytime you want to update its position, you can use ```line_np.node().moveTo()``` and ```line_np.node().drawTo()```. 

Besides, you can add some marks for showing a trajectory. In the following code segment, it will create 10 boxes for you to represent the trajectory
https://github.com/metadriverse/metadrive/blob/8f651ba1cdb9b5d702a0def802d8a41dfbd2419f/metadrive/component/vehicle_navigation_module/trajectory_navigation.py#L56

And each time, when your trajectory is updated, you can sync that information with the following code:
https://github.com/metadriverse/metadrive/blob/8f651ba1cdb9b5d702a0def802d8a41dfbd2419f/metadrive/component/vehicle_navigation_module/trajectory_navigation.py#L129

closed 2023-09-06 by QuanyiLi

CarlDegio, 2023-11-07:
Hello, I would like to know how long the time in a step metadrive represents (dt in every step)? I may need to refer to it when designing the trajectory.

pengzhenghao, 2023-11-07:
The `decision_repeat` is set to 5:

https://github.com/metadriverse/metadrive/blob/0d437097399b0b5cb7cde32880da30673eb8b435/metadrive/envs/base_env.py#L48

The  `physics_world_step_size` is set to 0.02s:

https://github.com/metadriverse/metadrive/blob/0d437097399b0b5cb7cde32880da30673eb8b435/metadrive/envs/base_env.py#L178

Therefore for a RL agent, dt between two env.step is 0.1s in the virtual world.

CarlDegio, 2023-11-07:
Thank you! I got that.

CarlDegio, 2023-11-29:
In order to develop a controller as good as possible, I would like to know where to get a dynamic model of the vehicle? I saw that there is a parameter to control whether to use random dynamics parameters, but I still don't understand what kind of vehicle dynamics model metadrive uses.

QuanyiLi, 2023-11-30:
MetaDrive vehicle uses BulletRayCastVehicle as the dynamic model, which moves the car by setting rotation on the four wheels just like a real vehicle. I am afraid you can not get the dynamics model in analytic expression. However, it is ok to use a bicycle model to approximate it with well-tuned parameters. One of my team members did this before and it indeed works with an MPC controller.

CarlDegio, 2023-12-09:
I found a problem. When `random_agent_model` in config is set False, agent_manager will spawn car of DefaultVehicle in `metadrive/component/vehicle/vehicle_type.py`. Then, it use VehicleParameterSpace.DEFAULT_VEHICLE as action parameters. But I found VehicleParameterSpace in `metadrive/component/pg_space.py` seems use random parameters? 
```
BASE_VEHICLE = dict(
        wheel_friction=ConstantSpace(0.9),
        max_engine_force=BoxSpace(750, 850),  # will sample from boxspace?
        max_brake_force=BoxSpace(80, 180),
        max_steering=ConstantSpace(40),
        max_speed_km_h=ConstantSpace(80),
    )
DEFAULT_VEHICLE = BASE_VEHICLE
```

I'm not sure if my understanding is correct, is it indeed using a random set of action parameters?