metadriverse/metadrive #451: [install bug] require shapely

CarlDegio, 2023-06-08:
```
python -m metadrive.examples.profile_metadrive
Traceback (most recent call last):
  File "<frozen runpy>", line 189, in _run_module_as_main
  File "<frozen runpy>", line 112, in _get_module_details
  File "D:\PythonSpace\metadrive\metadrive\__init__.py", line 1, in <module>
    import metadrive.register
  File "D:\PythonSpace\metadrive\metadrive\register.py", line 4, in <module>
    from metadrive.envs import MetaDriveEnv
  File "D:\PythonSpace\metadrive\metadrive\envs\__init__.py", line 1, in <module>
    from metadrive.envs.marl_envs import MultiAgentMetaDrive, MultiAgentBottleneckEnv, MultiAgentTollgateEnv, \
  File "D:\PythonSpace\metadrive\metadrive\envs\marl_envs\__init__.py", line 1, in <module>
    from metadrive.envs.marl_envs.marl_bottleneck import MultiAgentBottleneckEnv
  File "D:\PythonSpace\metadrive\metadrive\envs\marl_envs\marl_bottleneck.py", line 1, in <module>
    from metadrive.component.map.pg_map import PGMap
  File "D:\PythonSpace\metadrive\metadrive\component\map\pg_map.py", line 8, in <module>
    from metadrive.component.algorithm.BIG import BigGenerateMethod, BIG
  File "D:\PythonSpace\metadrive\metadrive\component\algorithm\BIG.py", line 7, in <module>
    from metadrive.component.pgblock.first_block import FirstPGBlock
  File "D:\PythonSpace\metadrive\metadrive\component\pgblock\first_block.py", line 3, in <module>
    from metadrive.component.lane.straight_lane import StraightLane
  File "D:\PythonSpace\metadrive\metadrive\component\lane\straight_lane.py", line 6, in <module>
    from metadrive.component.lane.pg_lane import PGLane
  File "D:\PythonSpace\metadrive\metadrive\component\lane\pg_lane.py", line 2, in <module>
    from shapely import geometry
ModuleNotFoundError: No module named 'shapely'
```
or run env:
```
python -m metadrive.examples.drive_in_single_agent_env
Traceback (most recent call last):
  File "<frozen runpy>", line 189, in _run_module_as_main
  File "<frozen runpy>", line 112, in _get_module_details
  File "D:\PythonSpace\metadrive\metadrive\__init__.py", line 1, in <module>
    import metadrive.register
  File "D:\PythonSpace\metadrive\metadrive\register.py", line 4, in <module>
    from metadrive.envs import MetaDriveEnv
  File "D:\PythonSpace\metadrive\metadrive\envs\__init__.py", line 1, in <module>
    from metadrive.envs.marl_envs import MultiAgentMetaDrive, MultiAgentBottleneckEnv, MultiAgentTollgateEnv, \
  File "D:\PythonSpace\metadrive\metadrive\envs\marl_envs\__init__.py", line 1, in <module>
    from metadrive.envs.marl_envs.marl_bottleneck import MultiAgentBottleneckEnv
  File "D:\PythonSpace\metadrive\metadrive\envs\marl_envs\marl_bottleneck.py", line 1, in <module>
    from metadrive.component.map.pg_map import PGMap
  File "D:\PythonSpace\metadrive\metadrive\component\map\pg_map.py", line 8, in <module>
    from metadrive.component.algorithm.BIG import BigGenerateMethod, BIG
  File "D:\PythonSpace\metadrive\metadrive\component\algorithm\BIG.py", line 7, in <module>
    from metadrive.component.pgblock.first_block import FirstPGBlock
  File "D:\PythonSpace\metadrive\metadrive\component\pgblock\first_block.py", line 3, in <module>
    from metadrive.component.lane.straight_lane import StraightLane
  File "D:\PythonSpace\metadrive\metadrive\component\lane\straight_lane.py", line 6, in <module>
    from metadrive.component.lane.pg_lane import PGLane
  File "D:\PythonSpace\metadrive\metadrive\component\lane\pg_lane.py", line 2, in <module>
    from shapely import geometry
ModuleNotFoundError: No module named 'shapely'
```

CarlDegio, 2023-06-08:
geopandas also has this problem. Maybe add them in setup.py

CarlDegio, 2023-06-08:
Besides, I install metadrive in new env(python=3.11), use `pip install metadrive`, and error:
```
python -m metadrive.examples.drive_in_single_agent_env
D:\program\miniconda3\envs\metadrive\Lib\site-packages\pkg_resources\__init__.py:121: DeprecationWarning: pkg_resources is deprecated as an API
  warnings.warn("pkg_resources is deprecated as an API", DeprecationWarning)
D:\program\miniconda3\envs\metadrive\Lib\site-packages\pkg_resources\__init__.py:2870: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('mpl_toolkits')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
D:\program\miniconda3\envs\metadrive\Lib\site-packages\pkg_resources\__init__.py:2870: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('google')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
Known pipe types:
  wglGraphicsPipe
(all display modules loaded.)
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "D:\PythonSpace\metadrive\metadrive\examples\drive_in_single_agent_env.py", line 58, in <module>
    raise e
  File "D:\PythonSpace\metadrive\metadrive\examples\drive_in_single_agent_env.py", line 38, in <module>
    o = env.reset()
        ^^^^^^^^^^^
  File "D:\PythonSpace\metadrive\metadrive\envs\base_env.py", line 392, in reset
    self.lazy_init()  # it only works the first time when reset() is called to avoid the error when render
    ^^^^^^^^^^^^^^^^
  File "D:\PythonSpace\metadrive\metadrive\envs\base_env.py", line 279, in lazy_init
    engine = initialize_engine(self.config)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\PythonSpace\metadrive\metadrive\engine\engine_utils.py", line 12, in initialize_engine
    cls.singleton = cls(env_global_config)
                    ^^^^^^^^^^^^^^^^^^^^^^
  File "D:\PythonSpace\metadrive\metadrive\engine\base_engine.py", line 30, in __init__
    EngineCore.__init__(self, global_config)
  File "D:\PythonSpace\metadrive\metadrive\engine\core\engine_core.py", line 202, in __init__
    gltf.patch_loader(self.loader)
    ^^^^^^^^^^^^^^^^^
AttributeError: module 'gltf' has no attribute 'patch_loader'
```
reference this:https://github.com/pokepetter/ursina/issues/237, I reinstall panda3d-gltf==0.13 and it solved. Maybe need a version constraint

QuanyiLi, 2023-06-09:
Thank you for the feedback! I will fix this soon

closed 2023-08-22 by QuanyiLi