metadriverse/metadrive #454: [Bug] Top-down view needs flip y-axis

QuanyiLi, 2023-06-12:
As we unify the coordinate system now, the Pygame coordinate is left-hand, while the Panda coordinate is right-hand. ```pygame.transform.flip() may need to call for flipping everything```

QuanyiLi, 2023-10-20:
@pengzhenghao This issue is exactly what you have encountered recently. Calling `flip()` is a workaround but lower the efficiency. A good solution is to flip `y` axis by adding `-` to where the `position.y` is used in pygame API.

closed 2023-11-30 by QuanyiLi

pengzhenghao, 2023-11-30:
A even better solution is to have a unified `Position` class that inherently has three properties:

`p = Position(0.0, 0.0, 0.0)`
`p.panda3d_position`
`p.metadrive_position`
`p.pygame_position`

PR #552 (2023-11-30): Documentation completion
## What changes do you make in this PR?

- Provide a detailed explanation for MetaDrive
- Clean up legacy code including nuPlan, Argoverse, and so on.
- Slightly refactor map creation code

close #454 

It is a part for #425 

## Checklist

* [ ] I have merged the latest main branch into current branch.
* [ ] I have run `bash scripts/format.sh` before merging.
* Please use "squash and merge" mode.